roboticslab-uc3m/installation-guides #1: Problems intalling YARP ubuntu 14.04 (Trusty)

RaulFdzbis, 2017-01-10:
Hi all!

Right now im working in a clean computer with ubuntu 14.04. This means that i have to reinstall all the beautiful repos we have in roboticslab :D.

I open this issue because following the YARP installation [guide](https://github.com/roboticslab-uc3m/installation-guides/blob/develop/install_yarp.md), i had to solve some issues. Maybe i did something wrong, or maybe the guide needs to be updated.

In addition to the steps in the installation guide. I had to do the following steps:

- Install eigen3, as simple as:

```bash
sudo apt-get install libeigen3-dev
```
- Turn off CREATE_GUISS flag. When i tried to run cmake, an error related with Qt5 and that module was presented. The reason of this error was because CREATE_GUISS depends on Qt5 and Qt5 is disable in our installation. The solution was just to desactivate that module. The  cmake command i used, was something like this.

```bash
cmake .. -DCREATE_LIB_MATH=ON -DCREATE_GUIS=ON -DCREATE_OPTIONAL_CARRIERS=ON -DENABLE_yarpcar_mjpeg_carrier=ON -DYARP_USE_QT5=OFF -DCREATE_GUIS=OFF # YARP_USE_QT5 can induce errors as of 02/2016
```

PeterBowman, 2017-01-10:
YARP underwent some noticeable changes in its latest releases, one of such is having dropped default GSL linkage in favour of Eigen (see description of commit asrob-uc3m/robotDevastation@0fcd210). Apart from that, carrier option names had their suffix trimmed (e.g. `ENABLE_yarpcar_mjpeg_carrier` has been replaced by `ENABLE_yarpcar_mjpeg`).

jgvictores, 2017-01-10:
I think it's safe to include:
`sudo apt-get install libeigen3-dev`

Regarding GUIs, the idea is to activate GUIs but deactivate QT5 to force QT4. Could you copy+paste the error log?

RaulFdzbis, 2017-01-11:
Sure!

-- YARP Version: 2.3.68 (2.3.68+139-20170110.8+gita95e7ca)
-- Detecting required libraries
-- CMake modules directory: /home/raul/repos/yarp/cmake;/home/raul/repos/yarp/cmake/ycm-0.2/find-modules;/home/raul/repos/yarp/cmake/ycm-0.2/modules;/home/raul/repos/yarp/cmake/ycm-0.2/3rdparty;/home/raul/repos/yarp/cmake/ycm-0.2/cmake-next;/home/raul/repos/yarp/cmake/ycm-0.2/cmake-3.1/Modules
CMake Warning at cmake/YarpFindDependencies.cmake:95 (message):
  Qt5 (required to build QCustomPlot) not found.
Call Stack (most recent call first):
  cmake/YarpFindDependencies.cmake:354 (checkbuildandset_dependency)
  CMakeLists.txt:62 (include)

-- Checking GNUCXX version 3/4 to determine  OpenCV /opt/net/ path
-- - DBG OPENCV_INCLUDE_DIR_CVCAM=OPENCV_INCLUDE_DIR_CVCAM-NOTFOUND
-- - DBG OPENCV_INCLUDE_DIR_CVCAM=OPENCV_INCLUDE_DIR_CVCAM-NOTFOUND
-- OPENCV_LIBRARY not found turning off OPENCV_FOUND
-- OPENCV_CVAUX_LIBRARY not found turning off OPENCV_FOUND
-- OPENCV library or headers not found. Please search manually or set env. variable OPENCV_ROOT to guide search.
-- I have found the following libraries:
--  --- YCM: not found
--  +++ ACE: found (/usr/lib/libACE.so)
--  +++ SQLite: compiling (not found)
--  +++ Eigen3: found (/usr/include/eigen3)
--  --- GSL: disabled
--  +++ TinyXML (>= 2.6): compiling (not found)
--  +++ xmlrpcpp: compiling (not found)
--  --- Qt5: not found
--  --- QCustomPlot: not found
--  --- Libedit: not found
--  --- SWIG: disabled
--  --- OpenCV: not found
--  --- Lua: not found
--  --- LibOVR: not found
--  --- GLFW3: not found
--  --- GLEW: not found
--  --- Libdc1394 (>= 2.0): not found
--  +++ JPEG: found (/usr/lib/x86_64-linux-gnu/libjpeg.so)
--  +++ MPI: found (/usr/lib/libmpi_cxx.so)
--  --- FTDI: not found
--  --- CUDA: not found
--  --- FFMPEG: not found
--  --- wxWidgets: not found
--  --- SDL: not found
--  --- PortAudio: not found
--  --- NVIDIACg: not found
--  --- Libusb1: not found
--  --- Stage: not found
CMake Error at cmake/YarpFindDependencies.cmake:161 (message):
  Optional package Qt5 not found.  Please install it or disable the option
  "CREATE_GUIS" to build yarp.
Call Stack (most recent call first):
  cmake/YarpFindDependencies.cmake:468 (check_optional_dependency)
  CMakeLists.txt:62 (include)

-- Configuring incomplete, errors occurred!

jgvictores, 2017-01-11:
I'm checking my install on Ubuntu 14.04.03 Trusty and I'm actually using Qt5. In fact, I haven't updated YARP very recently, but I can't even find `YARP_USE_QT5`.
So I'd just go on considering YARP+Qt5 is stable enough. Actions to be taken (in a new branch, for trying it out):
- Remove `-DYARP_USE_QT5=OFF` from our install guides.
- Include all required packages for Qt5. I may be wrong, but I recall there there a quite few. We can comment on them through this thread as you're the one with the clean machine. :smile:

RaulFdzbis, 2017-01-12:
Finally get this thing working thanks to @jgvictores.

What i had to do was to install the needed package as explained in the Yarp [homepage](http://www.yarp.it/dependencies.html#dependencies_qt5). This means:
```bash
sudo apt-get install qtbase5-dev qtdeclarative5-dev qtmultimedia5-dev \
  qml-module-qtquick2 qml-module-qtquick-window2 \
  qml-module-qtmultimedia qml-module-qtquick-dialogs \
  qml-module-qtquick-controls libqt5svg5
```

Note: In my case some of the packages were not found.

I also have had installed Qtcreator app from the official [web page](https://www.qt.io/download-open-source/). However i dont think this had any influence.

jgvictores, 2017-01-18:
Everything pretty decent as of e83d03f4a59e88009b0ce40d6c8e373f83c8c579, closing issue!

closed 2017-01-18 by jgvictores