roboticslab-uc3m/installation-guides #3: Openrave Dependencies

RaulFdzbis, 2017-02-13:
I had to install some additional dependencies than the ones presented in the [guide](https://github.com/roboticslab-uc3m/installation-guides/blob/develop/install_openrave.md). These were realted to the intallation of FCL:

Install **libccd** (dependency of FCL):

```bash
sudo add-apt-repository ppa:libccd-debs/ppa
sudo apt-get update
sudo apt-get install libccd-dev
```

Then install [FCL](https://github.com/flexible-collision-library/fcl) from source. (NOTE: The latest FCL version is not [working properly](https://bitbucket.org/ompl/ompl/issues/291/omplapp-installation-problem), I installed the 0.5 version instead).

jgvictores, 2017-02-14:
Hi!
1. What distro are you on?
2. Right now we indicate the use of the precompiled `libfcl-dev` package on the [guide](https://github.com/roboticslab-uc3m/installation-guides/blob/develop/install_openrave.md). Could you check if this package allows OpenRAVE to compile with FCL?
3. In case the previous fails, would you be okay with `-DOPENRAVE_PLUGIN_FCLRAVE=OFF` forced in th guide?

PeterBowman, 2017-02-14:
@RaulFdzbis I found these guides to be quite helpful:

* http://fsuarez6.github.io/blog/openrave-trusty/ (trusty, see next link if still in trouble with FCL)
* http://fsuarez6.github.io/blog/workstation-setup-xenial/ (xenial)

RaulFdzbis, 2017-02-14:
Hi! 

@PeterBowman  thank you for the guides! but installing the things mentioned it worked for me, so no problem there :D.

@jgvictores sorry for the lack of information :(

1. Im working with Linux Mint 17.3 Rosa
2. Maybe that was the problem, in my distro that package is not found.
3. In all the simulations i performed, it looked like FCL was not used at all (there was no collisions between objects). However i am not quite sure, what is the complete role that FCL performs inside the Openrave architecture, i think it should be tested first.

jgvictores, 2017-02-15:
Given the `Linux Mint 17.3 Rosa <-> Ubuntu Trusty` equivalence, could you try directly [installing from PPA](https://github.com/roboticslab-uc3m/installation-guides/blob/develop/install_openrave.md#install-openrave-ubuntu-1404)?

RaulFdzbis, 2017-02-15:
I tried, but it gives me "E: Unable to locate package openrave"

jgvictores, 2017-02-20:
Ok, my bad. The PPA for Trusty only offers collada packages, not openrave. Edited out on 0c4613c3ef04bae95d95faaf01d90cc03c8d8d84.

jgvictores, 2017-03-10:
Actions regarding FCL (all in [install_openrave.md](https://github.com/roboticslab-uc3m/installation-guides/blob/f7008537457e6c6dc60250ee19c38c0052e96532/install_openrave.md) and implemented in https://github.com/roboticslab-uc3m/installation-guides/commit/f7008537457e6c6dc60250ee19c38c0052e96532:
- Default to `-DOPENRAVE_PLUGIN_FCLRAVE=OFF`
- Document [Install OpenRAVE with FCL (Confirmed for Ubuntu 15.04, 15.10, 16.10, 17.04)](https://github.com/roboticslab-uc3m/installation-guides/blob/f7008537457e6c6dc60250ee19c38c0052e96532/install_openrave.md#install-openrave-with-fcl-confirmed-for-ubuntu-1504-1510-1610-1704)
- Link to this issue for further reference regarding compiling FCL from source.

closed 2017-03-10 by jgvictores