roboticslab-uc3m/installation-guides #7: Move to kebab-case for file names?

jgvictores, 2017-03-10:
Move to kebab-case for file names?

David-Estevez, 2017-03-13:
Gitbook automatically generates files with this convention if using the online/local Gitbook editor. For gitbooks, I would choose kebab-case as it prevents us for accidentally having files with a different convention due to the editor.

PeterBowman, 2017-03-13:
> Gitbook automatically generates files with this convention if using the online/local Gitbook editor. For gitbooks, I would choose kebab-case as it prevents us for accidentally having files with a different convention due to the editor.

+1 for kebab-case, then. I assume this issue also covers filenames used outside this repo, like [yarp-devices
/yarp_devices_install_on_debian_6.md](https://github.com/roboticslab-uc3m/yarp-devices/tree/76b7744/doc), manual_software.doc, etc.?

David-Estevez, 2017-03-13:
Are those related to a gitbook?

PeterBowman, 2017-03-13:
Hm, no, but they are related to roboticslab-uc3m. Which leads me to think that my previous assumption was incorrect, sorry.

jgvictores, 2017-03-13:
@David-Estevez @PeterBowman You are actually both right. Let's do this on both levels:
1. At organization level, for all files contained in `/doc/`. For this task I've opened: https://github.com/roboticslab-uc3m/best-practices/issues/7
2. Files contained in this repo ([installation-guides](https://github.com/roboticslab-uc3m/installation-guides)), which were in fact originally from `/doc/` of different repositories.

As (1) has been included at organization level, we'll close this issue once (2) is done.

jgvictores, 2017-03-15:
(1) Will be done at organization level.
(2) Done at https://github.com/roboticslab-uc3m/installation-guides/commit/2b16a4f42e740055e5cc4e506a7350c4db63b9d9

closed 2017-03-15 by jgvictores