roboticslab-uc3m/installation-guides #8: Replace `make`arbitrary -j parameter

jgvictores, 2017-03-10:
There are more sophisticated solutions, such as [[ref](https://www.math-linux.com/linux/tip-of-the-day/article/speedup-gnu-make-build-and-compilation-process)]:
```bash
NB_CORES=$(grep -c '^processor' /proc/cpuinfo)
make -j$((NB_CORES+1)) -l${NB_CORES}
```

PeterBowman, 2017-03-23:
Remaining `make -j` occurrences:

```
install-speech-recognition.md
install-kdl.md
install-icub.md
install-travislib.md
install-aquila.md
install-openrave.md
```

Retrieved with: `grep -Prl 'make \-j(?!\$)' --exclude-dir=.git`.

jgvictores, 2017-03-23:
Does the regex you cite include "orphan" makes? (no -j,, should be updated too...).

PeterBowman, 2017-03-23:
Hm, no, just looked at the title and forgot about all the plain `make`s around there. BTW doesn't it feel like too much clutter for such a simple task (compared to `-j$(nproc)`)?

jgvictores, 2017-03-23:
Maybe... or maybe... `NPROC=$(nproc); make -j$((NPROC+1)) -l${NPROC}` ??

PeterBowman, 2017-03-23:
IMO the fewer instructions the better - I'd remove `NPROC=$(nproc);` and inline `nproc` in following `make` calls.

jgvictores, 2017-03-23:
Yes, i agree. I have not found relevant sources regarding the benefits on -j(n+1) anyway [[1](http://unix.stackexchange.com/questions/208568/how-to-determine-the-maximum-number-to-pass-to-make-j-option)][[2](http://stackoverflow.com/questions/2499070/gnu-make-should-the-number-of-jobs-equal-the-number-of-cpu-cores-in-a-system)], so:
```
make -j$(nproc)
```
Looks good to me!

jgvictores, 2017-04-04:
Done at a03df57603d6686c90d0490a42131e8c87a38729.

closed 2017-04-04 by jgvictores