roboticslab-uc3m/installation-guides #13: Should we skip ace on yarp for linux?

jgvictores, 2017-04-09:
I'd like to unify what we do in travis-ci and what we recommend here. Should we skip ace on [install-yarp](https://github.com/roboticslab-uc3m/installation-guides/blob/develop/install-yarp.md) for linux (or alternatively, download ace on all travis builds)?

jgvictores, 2017-05-03:
After commenting with @PeterBowman, we will add `libace-dev` and remove `-DSKIP_ACE=ON` throughout `.travis.yml` (e.g. [here](https://github.com/roboticslab-uc3m/vision/blob/09d99acb61fe38efbe52d5cf468c093e7b90a007/.travis.yml#L46)).

PeterBowman, 2018-03-20:
On UNIX, ACE-less builds lead to the use of `dlopen`, which seems to choke on non-installed usage of YARP (which, in turn, involves playing with the `YARP_DATA_DIRS` and `LD_LIBRARY_PATH` environment variables, most notably the latter). Relevant lines: [ref](https://github.com/robotology/yarp/blob/29e9108a52713dbe737d9a9c44a62da3adc3ce51/src/libYARP_OS/src/SharedLibrary.cpp#L72-L95). Typical error on `yarpdev --list` with custom YARP devices: `[WARNING]Wrong library name for plugin blah` ([Travis](https://travis-ci.org/roboticslab-uc3m/kinematics-dynamics/builds/355352586)).

> After commenting with @PeterBowman, we will add libace-dev and remove -DSKIP_ACE=ON throughout .travis.yml (e.g. here).

I can't recall the exact reason for proceeding with this (probably unification of YARP's install guides with ours). Given that this issue is quite old now, are we still enforcing the ACE approach?

PeterBowman, 2018-03-20:
Con: there is no `libace-dev` at https://github.com/travis-ci/apt-package-whitelist in case we ever want to switch to containder-based CI builds (I planned to test it soon).

jgvictores, 2018-03-21:
Added the `question` label again, because I guess it's not clear...

My main concern is having the same recommended installation and Travis CI setup.

PeterBowman, 2018-03-22:
Note to self: check this on newer YARP versions (`yarp::os::impl::dlopen` vs `::dlopen`).

PeterBowman, 2018-04-06:
> On UNIX, ACE-less builds lead to the use of dlopen, which seems to choke on non-installed usage of YARP

Solved at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/ebcfb55df254d68ebbd3df28bbbdd7952d845a76 (see https://github.com/roboticslab-uc3m/questions-and-answers/issues/48#issuecomment-379345051). Just had to tweak the `YARP_DATA_DIRS` environment variable.

PeterBowman, 2018-06-30:
Marking as blocked by https://github.com/roboticslab-uc3m/questions-and-answers/issues/48, but we have already concluded that dropping the ACE dependency is the way to go. YAML script files for Travis will be updated accordingly as a result of that issue.

See also https://github.com/robotology/yarp/issues/1637 (full no-ACE support in future YARP).

closed 2018-07-13 by PeterBowman