roboticslab-uc3m/installation-guides #14: CUDA library not found when compiling

olayasturias, 2017-06-15:
OpenCV version: 3.0.0
O.S. : Ubuntu 14.04
**CUDA toolkit installed but without NVIDIA driver**

When compiling OpenCV 3.0.0 branch, the configure step fails with:

```bash
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_CUDA_LIBRARY (ADVANCED)
    linked by target "example_gpu_stereo_match" in directory /opencv/samples/gpu
    linked by target "example_gpu_multi" in directory /opencv/samples/gpu
    linked by target "example_gpu_driver_api_multi" in directory /opencv/samples/gpu
    linked by target "example_gpu_pyrlk_optical_flow" in directory /opencv/samples/gpu
    linked by target "example_gpu_optical_flow" in directory /opencv/samples/gpu
    linked by target "example_gpu_houghlines" in directory /opencv/samples/gpu
    linked by target "example_gpu_hog" in directory /opencv/samples/gpu
    linked by target "example_gpu_surf_keypoint_matcher" in directory /opencv/samples/gpu
    linked by target "example_gpu_stereo_multi" in directory /opencv/samples/gpu
    linked by target "example_gpu_video_reader" in directory /opencv/samples/gpu
    linked by target "example_gpu_morphology" in directory /opencv/samples/gpu
    linked by target "example_gpu_opengl" in directory /opencv/samples/gpu
    linked by target "example_gpu_cascadeclassifier" in directory /opencv/samples/gpu
    linked by target "example_gpu_alpha_comp" in directory /opencv/samples/gpu
    linked by target "example_gpu_generalized_hough" in directory //opencv/samples/gpu
    linked by target "example_gpu_opticalflow_nvidia_api" in directory /opencv/samples/gpu
    linked by target "example_gpu_video_writer" in directory /opencv/samples/gpu
    linked by target "example_gpu_super_resolution" in directory /opencv/samples/gpu
    linked by target "example_gpu_cascadeclassifier_nvidia_api" in directory /opencv/samples/gpu
    linked by target "example_gpu_farneback_optical_flow" in directory /opencv/samples/gpu
    linked by target "example_gpu_driver_api_stereo_multi" in directory /opencv/samples/gpu
    linked by target "example_gpu_bgfg_segm" in directory /opencv/samples/gpu
```

olayasturias, 2017-06-15:
Seems like it didn't found the libcuda.so

Solved by adding the following cmake flag:
`-DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs`

RaulFdzbis, 2017-06-15:
It looks like its your first issue, but good job!

David-Estevez, 2017-06-15:
Fix documented [here](https://roboticslab-uc3m.gitbooks.io/installation-guides/content/install-opencv.html). 

Let me know in case I need to add any additional steps that you might have done.

closed 2017-06-15 by David-Estevez