roboticslab-uc3m/installation-guides #15: Error when compiling cuda 8.0 library

olayasturias, 2017-06-15:
OpenCV version: 3.0.0
O.S. : Ubuntu 14.04
**CUDA toolkit installed but without NVIDIA driver**

OpenCV 3.0.0 shows various errors when compiling cuda 8.0, which look like this:
```bash
opencv/modules/cudalegacy/src/graphcuts.cpp:135:18: error: expected type-specifier before ‘NppiGraphcutState’
         operator NppiGraphcutState*()
opencv/modules/cudalegacy/src/graphcuts.cpp:141:9: error: ‘NppiGraphcutState’ does not name a type
         NppiGraphcutState* pState;
         ^
...
```

olayasturias, 2017-06-15:
Seems like it is solved in MASTER brach, but since I wanted to install the 3.0.0, i used the following patch:

```bash
git checkout 3.0.0
git format-patch -1 10896129b39655e19e4e7c529153cb5c2191a1db
git am < 0001-GraphCut-deprecated-in-CUDA-7.5-and-removed-in-8.0.patch
```

And it works now!

[image]

RaulFdzbis, 2017-06-15:
It looks like its your second issue, but good job!

David-Estevez, 2017-06-15:
Fix documented [here](https://roboticslab-uc3m.gitbooks.io/installation-guides/content/install-opencv.html). 

Let me know in case I need to add any additional steps that you might have done.

closed 2017-06-15 by David-Estevez