roboticslab-uc3m/installation-guides #16: Allow the specification of the program version

Kuronogard, 2017-09-08:
The current last version of YARP (2.3.70) seems to have problems when building the python bindings. Because of that, we have switched to 2.3.66.1 for the TIAGo proyect.

May be interesting (whenever possible) to add some step in the installation guides to specify the version to install, so they are more generic and can be included in projects with this kind of specific dependencies.

jgvictores, 2017-09-08:
There actually is a mechanism for this, see an example here:
https://github.com/roboticslab-uc3m/yarp-devices/blob/develop/doc/yarp_devices_install_on_debian_6.md

However, please note that this decision should be correctly justified, so please file an issue in this repo regarding the specific problems you're having, before pointing to YARP 2.3.66.1.

Kuronogard, 2017-09-08:
Ok, I close this issue to open a more specific one, then.

closed 2017-09-08 by Kuronogard