roboticslab-uc3m/installation-guides #18: Include installation of YARP v2.3.66.1?

Kuronogard, 2017-09-09:
Here is the thing... 

When compiling YARP v2.3.70 (the current version at the time) with python bindings:

```bash
cmake ..  -DYARP_COMPILE_BINDINGS=ON -DCREATE_PYTHON=ON
make -j$(nproc)
```

I get the following error:
```bash
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:192: Warning 389: operator[] ignored (consider using %extend)
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:193: Warning 389: operator[] ignored (consider using %extend)
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:273: Warning 321: 'hash' conflicts with a built-in name in python
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/Portable.h:31: Error: Syntax error in input(3).
make[2]: *** [lib/python/yarpPYTHON_wrap.cxx] Error 1
make[1]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

I have opened an issue in their repository (https://github.com/robotology/yarp/issues/1341), but we have to work with 2.3.66.1 until it is fixed. Do you think we should include it in the YARP installation guide?

jgvictores, 2017-09-09:
As of https://github.com/roboticslab-uc3m/installation-guides/blob/f45c19affd973c97d518272e4911daa5809c22e9/install-yarp.md we `git clone https://github.com/robotology/yarp`, which should clone the default `master` branch. Is there any issue with `master`?

Kuronogard, 2017-09-09:
Yes, master doesn't build either. As stated in https://github.com/robotology/yarp/issues/1341, it seems the problem is related with Ubuntu 14.04 using an old SWIG version, which does not support some c++11 features. Because of that, YARP dropped the support for Ubuntu 14.04.

I think we have 3 options:
 - Update the related machines to Ubuntu 16.04 (a bit of a mess)
 - Try to install a newer SWIG version and include the corresponding guide
 - Include installation instructions for a YARP compatible version

What do you think?

jgvictores, 2017-09-09:
With TIAGo we're pretty tied to Ubuntu 14.04. I like your second option, how about giving a try to [swig backport 3.0.2-1ubuntu1~ubuntu14.04.1](https://packages.ubuntu.com/source/trusty-backports/swig) (we did something similar [here](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-cmake.md#install-cmake-289-debian-6010))?

Kuronogard, 2017-09-09:
Ok, it builds with swing3.0, which can be installed directly with `sudo apt-get install swig3.0`. I'll test that in the lab machines on Monday.

I'll close this issue, as it seems to be solved =)

closed 2017-09-09 by Kuronogard

jgvictores, 2017-09-09:
Ok, added note on this at #6.