roboticslab-uc3m/installation-guides #20: Include ROS support instructions in YARP installation guide

Kuronogard, 2017-09-10:
TIAGo-main requires to compile YARP with ROS support (and python bindings)
```bash
cmake .. -DCREATE_LIB_MATH=ON -DCREATE_GUIS=ON -DCREATE_OPTIONAL_CARRIERS=ON -DENABLE_yarpcar_mjpeg_carrier=ON -DENABLE_yarpcar_tcpros_carrier=ON -DENABLE_yarpcar_xmlrpc_carrier=ON -DYARP_COMPILE_BINDINGS=ON -DCREATE_PYTHON=ON
make -j$(nproc) # Compile
```

As ROS seems to be quite extended, we could change the main YARP guide to include ROS support... or add a new section. What do you think?

jgvictores, 2017-09-10:
I'd add an additional section, similar to:
https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-yarp.md#install-additional-yarp-device-openni2deviceserver-ubuntu

jgvictores, 2017-09-10:
PS: Different sections, one for ROS and one for Python bindings.

closed 2017-09-14 by Kuronogard