roboticslab-uc3m/installation-guides #21: Switch to master on openrave?

jgvictores, 2017-09-12:
`latest_stable` as in our installation guides, per Rosen Diankov, is [over 5 years old](https://github.com/rdiankov/openrave/issues/296#issuecomment-324598804)).

jgvictores, 2017-11-15:
@rsantos88 Okay, great, thanks for reviving this topic.

Note to self: Maybe the `pcrecpp.h` error is not present in `master` anymore?

PeterBowman, 2017-11-15:
> Note to self: Maybe the pcrecpp.h error is not present in master anymore?

Can't be sure now, but it probably isn't anymore according to the latest comments at rdiankov/openrave#296.

jgvictores, 2017-11-17:
Final comments on `latest_stable`:
- CMake fails if FCL is not present, solved via `cmake .. -DOPENRAVE_PLUGIN_FCLRAVE=OFF` or installing the [correct version](https://github.com/roboticslab-uc3m/installation-guides/blob/7e3bc85d6cac41de8d1d51e6d8aeec410056f387/install-openrave.md#install-openrave-with-fcl-confirmed-for-ubuntu-1504-1510-1610-1704).
- Compilation fails if `pcrecpp.h` is not present, solved via `libpcre++-dev` or `libpcre3-dev` package.

Switching to `master`, where FCL is not REQUIRED (but can be used if detected), and compilation does not fail if `pcrecpp.h` is not detected.

jgvictores, 2017-11-17:
Done via ebf6b5e787870ec41d034a5aa56fd6499b6d8825.

closed 2017-11-17 by jgvictores