roboticslab-uc3m/installation-guides #22: Use apt instead of apt-get?

PeterBowman, 2017-09-14:
As @jgvictores found out, `apt-get` is just a another layer below `apt` (see e.g. `apt-cache`). The latter provides color feedback during data fetch, installation, etc., which is so cool.

See also https://askubuntu.com/a/446484.

jgvictores, 2017-09-14:
I'm using `apt` by habit now, but are we aware if at least most popular Debian-based distros (Ubuntu/Mint/...) and Debian itself provide this command? How about in a bit out-of-date versions (aka Debian 6.0)?

PeterBowman, 2017-09-14:
Per [this](https://packages.ubuntu.com/search?keywords=apt&searchon=names), [this](https://packages.ubuntu.com/trusty/amd64/apt/filelist) and the link above, It's available since Ubuntu Trusty. Linux Mint, which I'm not familiar with at all, lists an `apt` backport since the Serena release ([ref](http://packages.linuxmint.com/search.php?release=any&section=any&keyword=apt)), but [this answer](https://superuser.com/a/505416) and [Linux Mint Releases](https://linuxmint.com/download_all.php) suggest that it may be available since Qiana (v17).

I'm afraid we are not so lucky on Debian Squeeze (6.0). There is an `admin/apt`  package on `main` for i386 ([pkgs](http://archive.debian.org/debian/dists/Debian-6.0/main/binary-i386/)):

```
Package: apt
Priority: important
Section: admin
Installed-Size: 6116
Maintainer: APT Development Team <deity@lists.debian.org>
Architecture: i386
Version: 0.8.10.3+squeeze1
Replaces: manpages-pl (<< 20060617-3~)
Provides: libapt-pkg4.10
Depends: libc6 (>= 2.3.4), libgcc1 (>= 1:4.1.1), libstdc++6 (>= 4.4.0), zlib1g (>= 1:1.2.2.3), debian-archive-keyring, gnupg
Suggests: aptitude | synaptic | wajig, dpkg-dev, apt-doc, bzip2, lzma, python-apt
Conflicts: python-apt (<< 0.7.93.2~)
Filename: pool/main/a/apt/apt_0.8.10.3+squeeze1_i386.deb
Size: 2171308
MD5sum: a823518b46a3e4efb68a703521d43256
SHA1: 802d335573141644f3d20d7cb087084c3ef5c3b5
SHA256: f5bb9476c290a755f3a2aabd91ef59dc7a73b9fc36ba018191e794f3ea82d76d
Description: Advanced front-end for dpkg
 This is Debian's next generation front-end for the dpkg package manager.
 It provides the apt-get utility and APT dselect method that provides a
 simpler, safer way to install and upgrade packages.
 .
 APT features complete installation ordering, multiple source capability
 and several other unique features, see the Users Guide in apt-doc.
Tag: admin::package-management, hardware::storage, hardware::storage:cd, interface::commandline, network::client, protocol::{ftp,http,ipv6}, role::program, suite::debian, use::downloading, use::searching, works-with::software:package
```

...but it doesn't provide an `apt` executable per *Contents-i386.gz* ([ref](http://archive.debian.org/debian/dists/Debian-6.0/)). However, I found these instead:

```
usr/lib/jvm/java-6-openjdk/bin/apt                           java/openjdk-6-jdk
usr/lib/jvm/java-6-sun-1.6.0.26/bin/apt                      non-free/java/sun-java6-jdk
```

Not sure what it is, though. No [backports](http://archive.debian.org/debian-backports/dists/squeeze-backports/) available unless I missed something.

PeterBowman, 2017-09-14:
Newer Debian distros started providing `usr/bin/apt` since Jessie: https://packages.debian.org/jessie/i386/apt/filelist.

jgvictores, 2017-09-14:
I guess we can update it everywhere except for Debain 6.0 (yarp-devices) and OpenRAVE on Ubuntu 12.04, then!

jgvictores, 2017-11-24:
Learning from [here](https://github.com/asrob-uc3m/tutoriales/blob/master/software/linux/bash.md#bucles-en-bash) and [this in general](https://github.com/asrob-uc3m/tutoriales/blob/master/software/linux/bash.md) we can learn to visualize all the appearances of `apt-get` with (note that this is recursive, does **not** need to be done inside final leave directories):
```bash
find -type f -exec grep --color=always apt-get {} +  # Will refer to this as [1]
```

Therefore, a brute force replacement would be (also recursive):
```bash
find -type f -exec sed -i 's/apt-get/apt/g' {} +  # Will refer to this as [2]
```

With [2] that should be enough. We could see that [1] returns nothing, and we'd need to replace `apt-get` by `apt`to see any results. We could also see this via `git diff`.

To close this issue, we however, in addition have to _manually_ restore the `apt-get` of the sections related to Debian 6.0 and Ubuntu 12.04.

PS: For repositories other than this one, I've opened https://github.com/roboticslab-uc3m/questions-and-answers/issues/37 .

rsantos88, 2017-11-27:
Done it [here](https://github.com/roboticslab-uc3m/installation-guides/pull/25):
* replacement of `apt-get` by `apt`  except in installation sections with Ubuntu 12.04 and debian 6.0
* sections with Ubuntu 12.04 and debian 6.0 (not changed): `install-cmake.md` , `install-yarp.md` and `install-openrave.md`

jgvictores, 2017-11-27:
Done at #25 !!

closed 2017-11-27 by jgvictores