roboticslab-uc3m/installation-guides #23: Yarp deprecated functions: setPositionMode() in class yarp::dev::IPositionControl

rsantos88, 2017-10-16:
There are some yarp functions used in some aplications like **follow-me** or **teo-self-presentation** that are deprecated (maybe because we had a old yarp version installed). 
Doing a clean installation and following the [yarp instructions](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-yarp.md#install-yarp-ubuntu), we can see this error when we try to install teo-self-presentation: 
```bash
[ 57%] Built target dialogueManager
[ 71%] Building CXX object programs/BodyExecution/CMakeFiles/bodyExecution.dir/BodyExecution.cpp.o
/home/raul/repos/teo-self-presentation/programs/BodyExecution/BodyExecution.cpp: In member function ‘virtual bool teo::BodyExecution::configure(yarp::os::ResourceFinder&)’:
/home/raul/repos/teo-self-presentation/programs/BodyExecution/BodyExecution.cpp:42:30: error: ‘class yarp::dev::IPositionControl’ has no member named ‘setPositionMode’
     leftArmIPositionControl->setPositionMode();
                              ^
/home/raul/repos/teo-self-presentation/programs/BodyExecution/BodyExecution.cpp:60:31: error: ‘class yarp::dev::IPositionControl’ has no member named ‘setPositionMode’
     rightArmIPositionControl->setPositionMode();
                               ^
/home/raul/repos/teo-self-presentation/programs/BodyExecution/BodyExecution.cpp:78:27: error: ‘class yarp::dev::IPositionControl’ has no member named ‘setPositionMode’
     headIPositionControl->setPositionMode();
                           ^
make[2]: *** [programs/BodyExecution/CMakeFiles/bodyExecution.dir/BodyExecution.cpp.o] Error 1
make[1]: *** [programs/BodyExecution/CMakeFiles/bodyExecution.dir/all] Error 2
make: *** [all] Error 2

```
This can be the last changes of [YARP 2.3.70](https://github.com/robotology/yarp/releases):

> Removed some old interfaces methods, deprecated since yarp 2.3.65.
i.e: setPositionDirectMode(), setPositionMode(), setVelocityMode(),
setOpenLoopMode(),setTorqueMode()`, etc.

so, we can continue using the same [YARP version for Debian 6.0.10](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-yarp.md#install-yarp-2368-debian-6010) in Ubuntu systems... what do you think?

jgvictores, 2017-10-16:
In the YARP version we recommend for Debian 6.0.10 ([e1221283abb1abc77619429bb4d9408cbb1cf6c8  # Corresponds to 2.3.68+181-20170203.11+gite122128](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-yarp.md#install-yarp-2368-debian-6010)) the new `IControlMode2` already exists, so it should be safe to upgrade.

jgvictores, 2017-10-16:
PS: What TEO uses already implements the `IControlMode2` interface at [the current `develop` commit](https://github.com/roboticslab-uc3m/yarp-devices/blob/e293a80cb997ce6f7fdf0a035e124a0d88a11215/libraries/BodyYarp/CanBusControlboard/CanBusControlboard.hpp#L59).

PeterBowman, 2017-10-16:
Related: roboticslab-uc3m/questions-and-answers#16.

> (...) the new IControlMode2 already exists, so it should be safe to upgrade applications like follow-me or teo-self-presentation.

Keep in mind that there is no `IControlMode2` support for bindings, yet. That is, you won't be able to switch modes from Python code at YARP 2.3.70+ (of course, you can make and send your own Bottle with `set icmd 0 vel` or the like, "by hand").

jgvictores, 2017-10-16:
Regarding bindings, it would be easy to update [yarp.i](https://github.com/robotology/yarp/blob/7037d0b0bff7959c1c2d918dbbf238c2cf0a5dac/bindings/yarp.i) and PR. Two notes on this:
1. There is an incoming PR that affects bindings, I'd wait for that.
1. I've been thinking for a while that making an inverted `view()` function would reduce the need to be modifying `yarp.i` for each interface.

rsantos88, 2017-10-17:
Ok, when I try to run teo-self-presentation, I find this error message:
```bash
||| clearing context
||| adding context [bodyExecution]
||| configuring
||| default config file specified as bodyExecution.ini
||| checking [/home/teo/repos/teo-self-presentation/build/bin/bodyExecution.ini] (pwd)
||| checking [/home/teo/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/roboticslab-speech/robots/default] (robot yarp.d)
||| checking [/usr/local/share/roboticslab-yarp-devices/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-configuration-files/robots/default] (robot yarp.d)
||| checking [/home/teo/.config/yarp/contexts/bodyExecution] (context YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/contexts/bodyExecution] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/bodyExecution] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/bodyExecution] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/bodyExecution] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/contexts/bodyExecution] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/bodyExecution] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/bodyExecution] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/bodyExecution] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/roboticslab-speech/contexts/bodyExecution] (context yarp.d)
||| checking [/usr/local/share/roboticslab-yarp-devices/contexts/bodyExecution] (context yarp.d)
||| checking [/usr/local/share/teo-configuration-files/contexts/bodyExecution] (context yarp.d)
||| checking [/home/teo/.config/yarp/bodyExecution.ini] (YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/bodyExecution.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/bodyExecution.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/bodyExecution.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/bodyExecution.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/bodyExecution.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/bodyExecution.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/bodyExecution.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/bodyExecution.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/roboticslab-speech/bodyExecution.ini] (yarp.d)
||| checking [/usr/local/share/roboticslab-yarp-devices/bodyExecution.ini] (yarp.d)
||| checking [/usr/local/share/teo-configuration-files/bodyExecution.ini] (yarp.d)
||| did not find bodyExecution.ini
yarp: cannot read from bodyExecution.ini
Run "./bodyExecution --help" for options.
./bodyExecution checking for yarp network... [ok]
--------------------------------------------------------------
yarp: Port /bodyExecution/teo/leftArm/rpc:o active at tcp://2.2.2.100:10019/
yarp: Port /bodyExecution/teo/leftArm/command:o active at tcp://2.2.2.100:10020/
yarp: Port /bodyExecution/teo/leftArm/stateExt:i active at tcp://2.2.2.100:10021/
yarp: Sending output from /bodyExecution/teo/leftArm/rpc:o to /teo/leftArm/rpc:i using tcp
yarp: Sending output from /bodyExecution/teo/leftArm/command:o to /teo/leftArm/command:i using udp
yarp: Receiving input from /teo/leftArm/stateExt:o to /bodyExecution/teo/leftArm/stateExt:i using udp
[ERROR]expecting protocol 1 9 0, but the device we are connecting to has protocol version 1 5 1
[ERROR] please update YARP or the remotecontrolboard implementation
[ERROR]checkProtocolVersion failed 
yarp: Removing output from /bodyExecution/teo/leftArm/rpc:o to /teo/leftArm/rpc:i
yarp: Removing output from /bodyExecution/teo/leftArm/command:o to /teo/leftArm/command:i
yarp: Removing input from /teo/leftArm/stateExt:o to /bodyExecution/teo/leftArm/stateExt:i
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
robot leftArm device not available.
[INFO]RFModule failed to open.
```
maybe the problem is the yarp version installed on **teo-manipulation** (YARP 2.3.68) I am trying to update the yarp version but I found this [issue](https://github.com/roboticslab-uc3m/installation-guides/issues/24)

PeterBowman, 2017-10-17:
> [ERROR]expecting protocol 1 9 0, but the device we are connecting to has protocol version 1 5 1

Looks like pre-2.3.70 YARP was using protocol version 1.5.1 (all the way down to YARP 2.3.65). You should install YARP 2.3.70+ on both ends: the PC you are running the presentation app on and the robot itself (teo-manipulation).

jgvictores, 2017-10-17:
That's correct... WIP in #24, now called "Updating recommended version of YARP for Debian 6.0".

jgvictores, 2017-10-21:
Blocked by #24 and therefore https://github.com/roboticslab-uc3m/yarp-devices/issues/118.

rsantos88, 2018-01-26:
unblocked by #24

rsantos88, 2018-01-26:
no more `setPositionMode()` required with `yarp::dev::IPositionControl`
see commit here https://github.com/roboticslab-uc3m/teo-self-presentation/commit/6ee98b9f6c312a8b6888931006bdf090a5594399

PeterBowman, 2018-01-27:
@rsantos88 I think it's actually a good practice to set an initial control mode prior to using the remote controlboard. You can't be sure how did previous executions end (perhaps velocity or torque mode).

jgvictores, 2018-01-27:
`IControlMode2` FTW! :tada:

jgvictores, 2018-01-28:
This is,
```
headIPositionControl->setPositionMode();
```
And similar should not be simply erased, but instead become something more like:
```
headIControlMode2-> [[set to position mode via IControlMode2 method ]]
```

rsantos88, 2018-01-29:
Ok, so I understand that now we should use `IControlMode2` to set the initial mode and them `IPositionControl2` to configure the parameters (speeds, acceleration..) and set the positions.

jgvictores, 2018-01-29:
Yes, precisely!

PeterBowman, 2018-01-29:
> Looks like pre-2.3.70 YARP was using protocol version 1.5.1 (all the way down to YARP 2.3.65). You should install YARP 2.3.70+ on both ends: the PC you are running the presentation app on and the robot itself (teo-manipulation).

I've never noticed the `ignoreProtocolCheck` config option in the `remote_controlboard` device ([ref](https://github.com/robotology/yarp/blob/9712fdec65a676247d9a297928ff5f4ad8286901/src/libYARP_dev/src/devices/RemoteControlBoard/RemoteControlBoard.cpp#L1136)). That is, YARP v2.3.70+ *might* connect to the server wrapper in the onboard CPU even if still running YARP v2.3.68.

jgvictores, 2018-01-29:
> I've never noticed the `ignoreProtocolCheck` config option in the `remote_controlboard` device [...]

That seems dangerous (refs: [1](https://www.youtube.com/watch?v=iQR8c7S67nA),[2](https://www.youtube.com/watch?v=EVr__5Addjw)).

rsantos88, 2018-01-30:
I've updated **teo-self-presentation** using the new interfaces as we have commented here.
You can see [here](https://github.com/roboticslab-uc3m/teo-self-presentation/commit/97a9df4e8b4a95f95fe873e987f0425daa434b8e) the changes. It works well but if you see something wrong or improvable, tell me.
This app has been updated using the new yarp version, removing deprecated function as you know.
If you want to update other apps like follow-me, please open other issue.
I'll close it :-)

closed 2018-01-30 by rsantos88

jgvictores, 2018-01-30:
Further improvements on current `fix-bodyexecution` branch, namely:
- https://github.com/roboticslab-uc3m/teo-self-presentation/commit/2befb190f90f19eeebf897ae8b4e58716ff57c6a
- https://github.com/roboticslab-uc3m/teo-self-presentation/commit/92dee302daf09d63244b0f4fd0a7b0a045e11e17
- https://github.com/roboticslab-uc3m/teo-self-presentation/commit/24d5ae1c6572bfbc2f5f8ae83a40ec5590c29f70

rsantos88, 2018-01-30:
Pull request for this improvements [here](https://github.com/roboticslab-uc3m/teo-self-presentation/pull/4)
Thanks @jgvictores !

jgvictores, 2018-01-30:
Perfect! Merged!!
Thanks a lot.