roboticslab-uc3m/installation-guides #24: CMake 3.0 or higher is required

rsantos88, 2017-10-17:
I am trying to reinstall and update yarp on **teo-manipulation** with the last version but I need cmake 3.0 or higher.
The actual version of cmake is 2.8.9:
```bash
CMake Error at CMakeLists.txt:5 (cmake_minimum_required):
  CMake 3.0 or higher is required.  You are running version 2.8.9

-- Configuring incomplete, errors occurred!
```
It's possible to update to cmake 3.0 with debian 6.0?

jgvictores, 2017-10-17:
For Debian 6.0 we recommend YARP 2.3.68+181-20170203.11+gite122128 see:
https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-yarp.md#install-yarp-2368-debian-6010

PeterBowman, 2017-10-17:
> It's possible to update to cmake 3.0 with debian 6.0?

Sure: `git clone` https://github.com/Kitware/CMake, `git checkout` your version of choice, compile and install :).

rsantos88, 2017-10-17:
@jgvictores yes, I know... please, read this [issue](https://github.com/roboticslab-uc3m/installation-guides/issues/23)

PeterBowman, 2017-10-17:
You'll probably need this since latest YARP requires C++11 features:

jgvictores, 2017-10-17:
Okay. Renaming this issue for traceability.
The original reason we froze YARP version for Debain was possibly: https://github.com/roboticslab-uc3m/yarp-devices/issues/106
I think this has been solved, let's work on this!

rsantos88, 2017-10-17:
@PeterBowman yes, that's true. To compile with C++11 I need gcc 4.7, because the actual version is 4.4.5 (I can see this with `g++ --version`). So, I've followed this steeps:
```bash
wget --no-check-certificate https://ftp.gnu.org/gnu/gcc/gcc-4.7.0/gcc-4.7.0.tar.gz
tar xzf gcc-4.7.0.tar.gz
cd gcc-4.7.0
./contrib/download_prerequisites
cd ..
mkdir objdir
cd objdir
$PWD/../gcc-4.7.0/configure --prefix=$HOME/GCC-4.7.0 --enable-languages=c,c++
make
```
ok, in the `make` steep, all the time, the bash is showing this information in a loop:
```bash
checking for default BUILD_CONFIG... bootstrap-debug
checking for bison... no
checking for byacc... no
checking for yacc... no
checking for bison... no
checking for gm4... no
checking for gnum4... no
checking for m4... m4
checking for flex... no
checking for lex... no
checking for flex... no
checking for makeinfo... makeinfo
checking for expect... no
checking for runtest... no
checking for ar... ar
checking for as... as
checking for dlltool... no
checking for ld... ld
checking for lipo... no
checking for nm... nm
checking for ranlib... ranlib
checking for strip... strip
checking for windres... no
checking for windmc... no
checking for objcopy... objcopy
checking for objdump... objdump
checking for readelf... readelf
checking for cc... cc
checking for c++... c++
checking for gcc... gcc
checking for gcj... no
checking for gfortran... no
checking for gccgo... no
checking for ar... no
checking for ar... ar
checking for as... no
checking for as... as
checking for dlltool... no
checking for dlltool... no
checking for ld... no
checking for ld... ld
checking for lipo... no
checking for lipo... no
checking for nm... no
checking for nm... nm
checking for objdump... no
checking for objdump... objdump
checking for ranlib... no
checking for ranlib... ranlib
checking for readelf... no
checking for readelf... readelf
checking for strip... no
checking for strip... strip
checking for windres... no
checking for windres... no
checking for windmc... no
checking for windmc... no
checking where to find the target ar... host tool
checking where to find the target as... host tool
checking where to find the target cc... just compiled
checking where to find the target c++... just compiled
checking where to find the target c++ for libstdc++... just compiled
checking where to find the target dlltool... host tool
checking where to find the target gcc... just compiled
checking where to find the target gcj... host tool
checking where to find the target gfortran... host tool
checking where to find the target gccgo... host tool
checking where to find the target ld... host tool
checking where to find the target lipo... host tool
checking where to find the target nm... host tool
checking where to find the target objdump... host tool
checking where to find the target ranlib... host tool
checking where to find the target readelf... host tool
checking where to find the target strip... host tool
checking where to find the target windres... host tool
checking where to find the target windmc... host tool
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether -fkeep-inline-functions is supported... yes
configure: creating ./config.status
CONFIG_FILES=Makefile CONFIG_HEADERS= /bin/bash ./config.status
config.status: creating Makefile
make: Warning: File `/home/teo/src/objdir/../gcc-4.7.0/config/bootstrap-debug.mk' has modification time 2.4e+08 s in the future
CONFIG_SHELL="/bin/bash" /bin/bash ./config.status --recheck
running CONFIG_SHELL=/bin/bash /bin/bash /home/teo/src/objdir/../gcc-4.7.0/configure --prefix=/home/teo/GCC-4.7.0 --enable-languages=c,c++,lto --no-create --no-recursion
checking build system type... i686-pc-linux-gnu
checking host system type... i686-pc-linux-gnu
checking target system type... i686-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /bin/sed
checking for gawk... no
checking for mawk... mawk
checking for libitm support... yes
checking for gcc... gcc
checking for C compiler default output file name... a.out
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for suffix of executables... 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for gnatbind... no
checking for gnatmake... no
checking whether compiler driver understands Ada... no
checking how to compare bootstrapped objects... cmp --ignore-initial=16 $$f1 $$f2
checking for objdir... .libs
checking for PWL_handle_timeout in -lpwl... no
checking for version 0.11 (revision 0 or later) of PPL... no

*** This configuration is not supported in the following subdirectories:
     gnattools target-libada target-libgfortran target-libgo target-libffi target-zlib target-libjava target-libobjc target-boehm-gc
    (Any other directories should still work fine.)

```
it never finish and all the time is repeating the same...
I need gcc 4.7 with c++11 to compile CMake 3.0...

jgvictores, 2017-10-17:
Here are some links:
- https://nosemaj.org/debian-gcc-4-7
- http://www.tenpay.tech/questions/184855/installing-gcc-4-8-on-debian

Regarding the title and concern of this issue... maybe we should look into https://github.com/roboticslab-uc3m/yarp-devices/issues/118 (Try hc-driver-4.9) before venturing into this?

rsantos88, 2017-10-18:
Hi,
I've followed this steeps:
* I added a source on /etc/apt/sources.list: `deb http://ftp.uk.debian.org/debian/ jessie main non-free contrib`
* sudo aptitude update
* sudo aptitude install gcc-4.8
And this is the operation to do. Confirm me if it's right to do it (remember that this operation is on teo-manipulation PC)
```bash
The following NEW packages will be installed:
  cpp-4.8{a} gcc-4.8 gcc-4.8-base{ab} gcc-4.9-base{ab} libasan0{a} libatomic1{a} libcloog-isl4{a} libgcc-4.8-dev{a} libgmp10{a} libisl10{a} libitm1{a} libmpc3{a} libquadmath0{a} multiarch-support{a} 
The following packages will be upgraded:
  binutils libc-bin libc-dev-bin libc6 libc6-dev{b} libgcc1{b} libgomp1{b} libmpfr4 locales make pkg-config 
11 packages upgraded, 14 newly installed, 0 to remove and 1162 not upgraded.
Need to get 29.4 MB of archives. After unpacking 53.8 MB will be used.
The following packages have unmet dependencies:
  gcc-4.9-base: Breaks: gcc-4.4-base (< 4.4.7) but 4.4.5-8 is installed.
                Breaks: gcj-4.4-base (< 4.4.6-9~) but 4.4.5-2 is installed.
  libgomp1: Breaks: gcc-4.3 (< 4.3.6-1) but 4.3.5-4 is installed.
            Breaks: gcc-4.4 (< 4.4.6-4) but 4.4.5-8 is installed.
  libc6-i686: PreDepends: libc6 (= 2.11.3-4) but 2.19-18+deb8u10 is to be installed.
  libgcc1: Breaks: gcc-4.3 (< 4.3.6-1) but 4.3.5-4 is installed.
           Breaks: gcc-4.4 (< 4.4.6-4) but 4.4.5-8 is installed.
  gcc-4.8-base: Breaks: gcc-4.4-base (< 4.4.7) but 4.4.5-8 is installed.
                Breaks: gcj-4.4-base (< 4.4.6-9~) but 4.4.5-2 is installed.
  libc6-dev: Breaks: gcc-4.4 (< 4.4.6-4) but 4.4.5-8 is installed.
open: 63; closed: 698; defer: 15; conflict: 25                                                                                                                                                              .The following actions will resolve these dependencies:

      Remove the following packages:                                                                                                                  
1)      bsh                                                                                                                                           
2)      bsh-gcj                                                                                                                                       
3)      cpp-4.4                                                                                                                                       
4)      g++-4.4                                                                                                                                       
5)      gcc-4.3                                                                                                                                       
6)      gcc-4.4                                                                                                                                       
7)      gcc-4.4-base                                                                                                                                  
8)      gcj-4.4-base                                                                                                                                  
9)      gcj-4.4-jre                                                                                                                                   
10)     gcj-4.4-jre-headless                                                                                                                          
11)     gcj-4.4-jre-lib                                                                                                                               
12)     gcj-jre                                                                                                                                       
13)     gcj-jre-headless                                                                                                                              
14)     libdb4.7-java-gcj                                                                                                                             
15)     libgcj-bc                                                                                                                                     
16)     libgcj10                                                                                                                                      
17)     libgcj10-awt                                                                                                                                  
18)     libhsqldb-java-gcj                                                                                                                            
19)     libstdc++6-4.4-dev                                                                                                                            
20)     linux-headers-2.6-486                                                                                                                         
21)     linux-headers-2.6.32-5-486                                                                                                                    
22)     linux-headers-2.6.32-5-686                                                                                                                    
[... cut ...]
66)     libdb4.7-java recommends libdb4.7-java-gcj                                                                                                    
67)     libgcj10 recommends gcj-4.4-jre-lib (>= 4.4.4-17)                                                                                             
68)     openoffice.org-writer recommends default-jre | gcj-jre | java-gcj-compat | openjdk-6-jre | sun-java5-jre | sun-java6-jre | java5-runtime | jre
69)     liblucene2-java recommends libdb-java (>= 5.1.4)   
   

Accept this solution? [Y/n/q/?]      
```

rsantos88, 2017-10-18:
Waiting for new moves with roboticslab-uc3m/yarp-devices#118 (Try hc-driver-4.9), to continue working with TEO, all computers have to have the same YARP version (2.3.68)

rsantos88, 2018-01-25:
this issue is no longer blocked thanks to https://github.com/roboticslab-uc3m/teo-main/issues/26

rsantos88, 2018-01-25:
Using Ubuntu 14.04, the yarp version used now is the last version `YARP version 2.3.70.2` in both computers (client and Teo computers)
I think we can close this issue

rsantos88, 2018-01-26:
this issue will not block #23 anymore thanks to https://github.com/roboticslab-uc3m/yarp-devices/issues/118
I'll close it :)

closed 2022-05-23 by PeterBowman