roboticslab-uc3m/installation-guides #26: Add section on YARP with Python3

jgvictores, 2017-11-30:
Some users have been struggling with YARP and Python3.

Here are some pointers to issues that may help for adding a section to the guide, or at least help for now:
- Most recent: https://github.com/roboticslab-uc3m/yarp-devices/issues/145#issuecomment-348230791
- https://github.com/robotology/yarp/issues/1341
- https://github.com/robotology/yarp/issues/1024

gsaponaro, 2018-04-23:
Hi @jgvictores , I was having an "outsider's view" on your guides. Pretty useful stuff, thank you. About this particular issue #26, I built YARP Python 3 bindings like this: https://github.com/robotology/yarp/issues/1149#issuecomment-355042530
My system has both Python 2 and Python 3. If I didn't do the above trick, YARP would give priority to Python 2.

jgvictores, 2018-04-24:
> Pretty useful stuff, thank you. 

Thanks! We actually had a small debate at https://github.com/roboticslab-uc3m/questions-and-answers/issues/3 before setting it up. I'm usually in favor of improving external projects rather than storing internal documents, but sometimes PRs are too slow, or we may want to do specific stuff like freeze at a certain version.

>  I built YARP Python 3 bindings like this: robotology/yarp#1149 (comment)

That's helpful and also reactivates the topic. :-)

I'm still bumping into something like [this](https://github.com/roboticslab-uc3m/yarp-devices/issues/145#issuecomment-348230791) (`CMAKE_INSTALL_PYTHONDIR` is apparently ignored), which I've documented at 772afc4. I wanted to dive a bit more into the CMake before posting yet another issue on YARP, but feel free to open it, we could continue the conversation there if you consider it noteworthy.

Cheers!

jgvictores, 2018-04-24:
PS: Issue not exactly the same, there there was a `/usr` vs `/usr/local` issue... and comment on `PYTHON_INSTDIR` vs `CMAKE_INSTALL_PYTHONDIR`; still similar.

gsaponaro, 2018-04-24:
Sure, I'll be glad to help with YARP/Python/CMake tests. I'm not knowledgeable in SWIG though.

Not sure what kind of issue/patch should be proposed upstream at this moment. Feel free to tag me if you think of something. Ideally, on a system with the latest Ubuntu LTS (currently 16.04, soon 18.04) and both Python 2 and 3, I'd like to have the possibility of selecting between 2 and 3 in YARP CMake robustly.

PeterBowman, 2020-03-12:
See https://github.com/robotology/QA/issues/391.

PeterBowman, 2020-08-26:
YARP has moved on since 2017/18, now **Python 3 is required** and collisions between Python v2/v3 should no longer be a thing, if I'm correct. At least I managed to compile and use it successfully with just a tiny bit of hassle, properly reported and documented thereafter by myself at https://github.com/roboticslab-uc3m/installation-guides/commit/7a9134f9c1d39df4d980307c06c8fa6912dc9e62.

@jgvictores do you think that guide needs to expand a bit further on this matter, or is it good to go and you're fine for this issue to be closed?

jgvictores, 2020-08-27:
Good to go, thank you all!

closed 2020-08-27 by jgvictores