roboticslab-uc3m/installation-guides #33: [openrave] GCC build fails due to not finding mpfr.h

jgvictores, 2018-01-08:
Filed issue upstream, https://github.com/rdiankov/openrave/issues/543

PeterBowman, 2020-09-21:
MPFR is a hard dep to fparser, which is in turn a hard dep to OpenRAVE. If not found, OR tries to compile it from the sources located in the `extern/` directory. If found, you probably already have MPFR. I guess we need to add `sudo apt install libmpfr-dev` in the install guide.

jgvictores, 2020-09-23:
Or instead, in line with https://github.com/rdiankov/openrave/issues/543#issuecomment-358862523, disable it similar to [install-openrave.md (ATTOW)](https://github.com/roboticslab-uc3m/installation-guides/blob/d977c8f1a841283a8c6c59f7072a5f8f28f00d81/install-openrave.md)'s `cmake .. -DOPT_VIDEORECORDING=OFF  # Avoids AV errors`?

jgvictores, 2020-09-23:
Edited my previous comment, maybe https://github.com/rdiankov/openrave/issues/543#issuecomment-358862523 missed the hard-dep chain you mentioned. Then, yes.