roboticslab-uc3m/installation-guides #34: Remember to document RSA host key in any place...

rsantos88, 2018-01-19:
Error connecting by ssh to a new ubuntu distribution (or a new installation)
```bash
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the RSA key sent by the remote host is
94:43:6d:2b:78:54:06:7b:e5:02:61:01:d1:4a:35:51.
Please contact your system administrator.
Add correct host key in /home/teo/.ssh/known_hosts to get rid of this message.
Offending RSA key in /home/teo/.ssh/known_hosts:1
  remove with: ssh-keygen -f "/home/teo/.ssh/known_hosts" -R manipulation
RSA host key for manipulation has changed and you have requested strict checking.
Host key verification failed.
teo@oliver:~$ ssh manipulation 
teo@manipulation's password: 
Welcome to Ubuntu 14.04.5 LTS (GNU/Linux 4.4.0-31-generic i686)

 * Documentation:  https://help.ubuntu.com/
New release '16.04.3 LTS' available.
Run 'do-release-upgrade' to upgrade to it.

The programs included with the Ubuntu system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Ubuntu comes with ABSOLUTELY NO WARRANTY, to the extent permitted by
applicable law.
```

Remember to document steps to configure the ssh server (in case it was necessary) :
1. sudo apt-get install openssh-server
2. edit `/etc/ssh/ssh_host_rsa_key` (with sudo)
3. change the old key for this:
```bash
-----BEGIN RSA PRIVATE KEY-----
MIIEpAIBAAKCAQEA2c9dUMx8x0V12GUxg4qGH0T+m+SE/+xvePCGkBJNiMWJGAW4
QpzYnYirdSmNNNrs0EhDfUinMlhBFSkDvqBcGNOalmLOIK17kssMPBnihj7YQJub
L/YhS1e8rjF0xKDJFiJ/hHrAXw/JhYg9rclmcx+snDZBqfIM13UYg/wfST6GYtVF
hwl6Przx3yoj6qVE48d5u3ggEOZIwaKFcJarpINEj5212g+rLmB8i7uCH7Nk32v5
ll70+0Z4wGemHSgvOw158HpfTr4HVLT2S/zAp/ESxgFceH0xBUOR4Ldc99bNZyzk
+p5V1o1vAbduFwCVt/WPqznUUEaTkEp2mGiNzwIDAQABAoIBAC3nWg9RSWrl5LQ6
f3jtlN66N767JpyZQ9UFPBbZBKYym3YInpDLZYut5hTHMQ9WpFoxIKgiy7n8h5ri
Dfv65+4DiqzvHeodJdIGIQn5mpgYXpMfOFGfWdz7HLTJj4ZxLHARzTYqeKiQNzSB
Rh5+N61Dm6lAdg5PCQ3Pi98lH6LburQ7xzUgsGOyoAcp6GvW+45XSj8dFCkosEjY
uLmeoEC/ndZkC6X/twFiQ0vTNTwOp6QJ6cA3f6gvwZ2MBM7KvN7e1mVPMEv7u7ml
eAlIWEzy6xJPirQHDPoxYZuo2Gaj28WNtNqTIbSQzN1l9ApbRwFAVC5ly7rDL5Fp
zEZXFGECgYEA/+HXrdXMlKxia9fh05Kjb2aud5QTdsf5bTJhdJxgrVIdtWWRw6zt
lG5hEARfng++B6Vq6QdMcg20zUvjcLXtV3LKHAhZGFJioq7OD9X+zT4sL966RlIy
sPeWtceKjPQvGRMdcdrLEpbSItggIEKrLg+yDdEFxZt6lxeeOQ7EhosCgYEA2ekI
8jLVcg6Nwh8egej1dsoTI/2XdaSeZHb+A/WozJYpYpHhNFgSNGKPUGQ+vB2HIDwg
GHPEJkLD0/wV2NON9VVtmi23c7TjEmcooT0e6cBNmOP3l2F+v8oXZn7Y9fuRtvie
ye9pwi33UrhUVJH+C2tVkgnDedNvpANcSbGdAk0CgYEAzhrnUsoHQymnDR6Sc4yF
wFGA7iiDlQV2TxJplTfwVyrxaw8o3egIZaqS3AfpzLL5acc/2ftbA/MuUr5xsHOZ
fPqk+PdwbDrP4CHVMMQ5vK/7f4NznvjOg+SprDnRy79TwGq5KOAG+i4pjI9xMQsC
PtO9o4tdjtGxaMpJU8QlYAkCgYA21fm2WSx1Y0wxCXu33VPH7NocOHig1ig1x+vs
OymbMj5/2438ih0qqZK+/Dr9Q8D5T6gT/y3hIvP8qHcm9TXIAxNttQtxRpfNzb/f
s23QDOB6FQP9SD+JK+XvmQQoAoKMtMvgrSrD/qU6X8B497QiU4RZF7PlOFvjviS/
Nr74EQKBgQCYFi3Bfw1Cxj+ebI7WiuKk4w9mEBM0Qfbu9VW2+8sphajbg45B7Zfc
zE5QGfPlKWNlTUrualruLnBwyI+z2pCdbWTjZnbw898+fv8H0hbBBNqZtfbLnjRp
7jNaspB23N02pepSujldWKjLVdVHpv3Lz3M+XSsUA3kaExTDBwduyA==
-----END RSA PRIVATE KEY-----
```

jgvictores, 2018-01-19:
ssh-keygen -f "/home/teo/.ssh/known_hosts" -R manipulation

<https://github.com/roboticslab-uc3m/installation-guides/issues/34>, or mute
the thread

PeterBowman, 2018-01-19:
(Private ssh keys are meant to be... private.)

AlvaroMartinezR, 2018-01-19:
@PeterBowman is right 
[image]

rsantos88, 2018-01-19:
:expressionless: :expressionless: :expressionless: :expressionless: :expressionless: :expressionless:
sorry for the error... where do you want to keep the key..? maybe in some private place for us...?

David-Estevez, 2018-01-19:
I guess the best thing to do is to create a new one (and keep it private) :wink:

jgvictores, 2018-01-20:
Obviously it's good to keep good security practices, but since we are in a LAN, I think we don't have to be extremely paranoid. Based on this, and to be practical, I think it's okay to not consider a man-in-the-middle attack and simply erase the RSA key associated to `manipulation` (via `ssh-keygen -f "/home/teo/.ssh/known_hosts" -R manipulation`) each time we boot it on a different distro.

Please correct me guys if I'm saying something terribly misguided.

rsantos88, 2018-01-22:
@jgvictores I am totally agree with you. I tried to change the default password assigned in Ubuntu for the one I had already registered in Debian 6 to avoid possible future problems. I know that to reveal a password in a public place like this is a mistake, but I thought that we are connected in a local network, free of dangerous hackers trying to get into Teo (or maybe yes..) and I am always trying to document all the steps or changes that are performed with the intention that another person can configure it in the future.
Anyways, I hope you have fun with this :sweat_smile:

rsantos88, 2018-01-23:
It has generated a new RSA key (reinstalling openssh-server) and now the solution is erase the RSA key associated to manipulation (via ssh-keygen -f "/home/teo/.ssh/known_hosts" -R manipulation) each time we boot it on a different distro (and the same with locomotion).
For example, doing ssh with locomotion:
```bash
teo@oliver:~$ ssh-keygen -f "/home/teo/.ssh/known_hosts" -R locomotion
# Host locomotion found: line 4 type RSA
/home/teo/.ssh/known_hosts updated.
Original contents retained as /home/teo/.ssh/known_hosts.old
teo@oliver:~$ ssh locomotion
The authenticity of host 'locomotion (2.2.2.52)' can't be established.
ECDSA key fingerprint is ??:??:??:??:??:??:??:??:??:??:??:??:??:??:??:??.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'locomotion' (ECDSA) to the list of known hosts.
Warning: the ECDSA host key for 'locomotion' differs from the key for the IP address '2.2.2.52'
Offending key for IP in /home/teo/.ssh/known_hosts:4
Are you sure you want to continue connecting (yes/no)? yes
teo@locomotion's password: 
Welcome to Ubuntu 14.04.5 LTS (GNU/Linux 4.4.0-31-generic i686)

 * Documentation:  https://help.ubuntu.com/

The programs included with the Ubuntu system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Ubuntu comes with ABSOLUTELY NO WARRANTY, to the extent permitted by
applicable law.

teo@locomotion:~$ 
```
So, I'll close this issue

closed 2018-01-23 by rsantos88