roboticslab-uc3m/installation-guides #38: [pcl] libpcl-dev gone from ppa in trusty

jgvictores, 2018-02-07:
Dispite http://pointclouds.org/downloads/linux.html, https://launchpad.net/~v-launchpad-jochen-sprickerhof-de/+archive/ubuntu/pcl is marked as deprecated in favor for official ubuntu packages.

This is true for xenial and above [ref](https://packages.ubuntu.com/search?keywords=libpcl-dev), but not for trusty: https://launchpad.net/~v-launchpad-jochen-sprickerhof-de/+archive/ubuntu/pcl?field.series_filter=trusty

Affects:
- https://github.com/roboticslab-uc3m/vision due to pointAtObjectServer which is currently disabled
- Whatever depends on above, e.g. https://github.com/roboticslab-uc3m/teo-main

jgvictores, 2018-02-07:
Looking at `pcl-1.7 - 1.7.2-1+trusty6` from https://launchpad.net/~v-launchpad-jochen-sprickerhof-de/+archive/ubuntu/pcl/+packages?batch=75&memo=75&start=75 perhaps `libpcl-1.7-all-dev` still exists in the jochen-sprickerhof PPA.

jgvictores, 2018-02-07:
A bit confusion... In vision it seems like travis was OK...

PeterBowman, 2018-11-18:
I'm going to mark this as invalid: the PCL packages are still there, Travis has been successful in the vision repo for months. Please reopen if any further action is needed.

closed 2022-05-23 by PeterBowman