roboticslab-uc3m/installation-guides #39: need to update cmake in this point

rsantos88, 2018-02-23:
Trying to install the [python bindings](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-yarp.md#install-python-bindings) it shows an error doing cmake ..
```bash
teo@teo-oliver:~/repos/yarp/build$ cmake ..
-- YARP Version: 2.3.72 (2.3.72+17-20180220.1+git8ae349c)
-- Using:
--   CMake: 3.5.1 (/usr/bin/cmake)
--   C Compiler: GNU 4.8.4 (/usr/bin/cc)
--   CXX Compiler: GNU 4.8.4 (/usr/bin/c++)
-- Detecting required libraries
-- CMake modules directory: /usr/local/share/YCM/modules;/usr/local/share/YCM/find-modules;/usr/local/share/YCM/build-modules;/usr/local/share/YCM/style-modules;/usr/local/share/YCM/cmake-next/Modules;/usr/local/share/YCM/cmake-3.7/Modules;/usr/local/share/YCM/3rdparty;/usr/local/share/YCM/deprecated;/home/teo/repos/yarp/cmake
-- I have found the following libraries:
--  +++ YCM (>= 0.6.0): found 0.6.0.1-20171221.2+git331f217 (/usr/local/share/YCM/cmake)
--  +++ ACE: found 6.0.3 (/usr/lib/libACE.so)
--  --- RTF (>= 1.4.0): not found
--  +++ SQLite: found 3.8.2 (/usr/lib/x86_64-linux-gnu/libsqlite3.so)
--  --- Eigen3: disabled
--  +++ TinyXML (>= 2.6): found 2.6.2 (/usr/lib/x86_64-linux-gnu/libtinyxml.so)
--  --- Qt5: disabled
--  --- QCustomPlot: disabled
--  --- Graphviz: disabled
--  +++ Libedit: found 3.1 (/usr/lib/x86_64-linux-gnu/libedit.so)
--  +++ SWIG (>= 3.0): found 3.0.2 (/usr/share/swig3.0)
--  +++ OpenCV: found 3.0.0 (/usr/local/share/OpenCV)
--  --- Lua: not found
--  --- LibOVR (>= 1.3): not found
--  --- GLFW3: not found
--  +++ GLEW: found (/usr/lib/x86_64-linux-gnu/libGLEW.so)
--  --- Libdc1394 (>= 2.0): not found
--  +++ JPEG: found (/usr/lib/x86_64-linux-gnu/libjpeg.so)
--  +++ MPI: found (/usr/lib/libmpi_cxx.so)
--  --- FTDI: not found
--  --- CUDA: not found
--  +++ FFMPEG: found (/usr/lib/x86_64-linux-gnu/libavformat.so)
--  --- wxWidgets: not found
--  --- SDL: not found
--  --- PortAudio: not found
--  --- NVIDIACg: not found
--  +++ Libusb1: found 1.0.17 (/usr/lib/x86_64-linux-gnu/libusb-1.0.so)
--  --- Stage: not found
--  --- ZFP (>= 0.5.1): not found
--  +++ OpenNI2: found 2.2.0.3 (/usr/local/lib/libOpenNI2.so)
--  +++ GLIB2: found (/usr/lib/x86_64-linux-gnu/libglib-2.0.so)
--  --- GStreamer (>= 1.4): not found
--  --- GStreamerPluginsBase (>= 1.4): not found
-- starting plugin library: yarpmod
--  --- plugin yarpmod_cuda: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_depthCamera: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_fakeDepthCamera: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_fakebot: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_fakeMotionControl: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_fakeAnalogSensor: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_fakeIMU: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_nvidia: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_stage: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_vfw_grabber: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_wxsdl: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_SerialServoBoard: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_ffmpeg_grabber: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_ffmpeg_writer: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_opencv_grabber: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_serial: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_serialport: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_portaudio: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_imuBosch_BNO055: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_jrkerr: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_urbtc: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_dynamixelAX12Ftdi: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_fakeLaser: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_laserFromDepth: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_rpLidar: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_rpLidar2: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_laserHokuyo: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_KinectDeviceLocal: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_KinectDeviceClient: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_OpenNI2DeviceServer: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_OpenNI2DeviceClient: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_ovrheadset: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_pumacalibrator: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_meiMotionControl: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_test_grabber: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_SDLJoypad: disabled (dependencies unsatisfied)
--  --- plugin yarpmod_microphone: disabled (dependencies unsatisfied)
-- ending plugin library: yarpmod
-- starting plugin library: yarpcar
--  --- plugin yarpcar_human: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_mpi: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_mpibcast: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_xmlrpc: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_tcpros: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_rossrv: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_mjpeg: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_bayer: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_priority: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_portmonitor: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_depthimage: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_zfp: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_h264: disabled (dependencies unsatisfied)
-- ending plugin library: yarpcar
--  --- bindings Java is not enabled
--  +++ bindings Python is enabled
--  --- bindings Perl is not enabled
--  --- bindings Chicken Scheme is not enabled
--  --- bindings CSharp is not enabled
--  --- bindings Allegro Common Lisp is not enabled
--  --- bindings TCL is not enabled
--  --- bindings Ruby is not enabled
--  --- bindings Lua is not enabled
--  --- bindings Octave is not enabled
CMake Error at bindings/python/CMakeLists.txt:33 (swig_add_library):
  Unknown CMake command "swig_add_library".

-- Configuring incomplete, errors occurred!
See also "/home/teo/repos/yarp/build/CMakeFiles/CMakeOutput.log".
See also "/home/teo/repos/yarp/build/CMakeFiles/CMakeError.log".
```
Looking for this problem, I found that we need to update cmake to a newer version.
My old version was `cmake version 3.5.1` but now I've installed the newest (cmake-3.11.0) and it works perfect.
Maybe it's necessary to document this... isn't it?

PeterBowman, 2018-02-23:
@jgvictores reported this upstream at https://github.com/robotology/yarp/issues/1552. It's a twofold problem:

* `swig_add_module` was renamed to `swig_add_library` in CMake 3.8, so it makes sense that an upgrade helps to solve this ([ref1](https://cmake.org/cmake/help/v3.8/module/UseSWIG.html), [ref2](https://cmake.org/cmake/help/v3.9/release/3.8.html#modules)).
* YCM is supposed to fetch modern CMake modules (i.e. the updated `UseSwig` at 3.8) but, per @jgvictores's comment, there is some shadowing happening there that seems to go away when you uninstall YCM (I can confirm this).

rsantos88, 2018-03-05:
I've added this note here https://github.com/roboticslab-uc3m/installation-guides/pull/40
Do the PR if you are agree :)
I'll close this issue.

jgvictores, 2018-03-05:
As commented ar #40, best option here is to uninstall YCM (could be reinstalled after compiling bindings) until issue is solved upstream.
Re-opening until robotology/yarp#1552 (or YCM-equivalent issue) is solved.

PeterBowman, 2018-06-24:
Solved in YCM 0.8.x with https://github.com/robotology/ycm/pull/139.

closed 2018-06-24 by PeterBowman