roboticslab-uc3m/installation-guides #44: `install-speech-recognition.md` is coupled.

jgvictores, 2018-04-25:
`install-speech-recognition.md` is coupled. Tasks:
- [ ] split into all the subcomponents
- [ ] Update https://github.com/roboticslab-uc3m/speech/blob/2ed13755d900721a139f300daf37ad185bc7fcf6/doc/speech-install.md
- [ ] Review if used elsewhere, if true: update

Alternative option for the lazy:
- [ ] Simply move into https://github.com/roboticslab-uc3m/speech/blob/2ed13755d900721a139f300daf37ad185bc7fcf6/doc/speech-install.md

rsantos88, 2018-05-08:
This files should be located in this repository? Because now are located in [speech/doc](https://github.com/roboticslab-uc3m/speech/tree/2ed13755d900721a139f300daf37ad185bc7fcf6/doc)

jgvictores, 2018-05-08:
Ideally, we would chop within this repo, then update links in speech.

jgvictores, 2018-05-09:
IMHO, still stuff to do... The following should make sense:
- https://github.com/roboticslab-uc3m/speech/blob/develop/doc/speech-install.md currently looks like [this](https://github.com/roboticslab-uc3m/speech/blob/651b91a6d3a8e4979f5a7b5467670b649b0aa292/doc/speech-install.md)
- https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-speech-recognition.md probably shouldn't even exist, and currently looks like [this](https://github.com/roboticslab-uc3m/installation-guides/blob/7bc24f0274414696e5b143e0aca8db6d5d6c3b14/install-speech-recognition.md)

rsantos88, 2018-05-09:
revised and corrected ;)

jgvictores, 2018-05-11:
Ok, trying to track changes.

The old: https://github.com/roboticslab-uc3m/installation-guides/blob/be39af91bc90afde698108257deb7bc2c0929407/install-speech-recognition.md

Is now at:
- https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-sphinxbase.md ([current](https://github.com/roboticslab-uc3m/installation-guides/blob/8f2f1a1a8a39f8bcc88cf07bbd483b1e1b8dd0da/install-sphinxbase.md))
- https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-pocketsphinx.md ([current](https://github.com/roboticslab-uc3m/installation-guides/blob/8f2f1a1a8a39f8bcc88cf07bbd483b1e1b8dd0da/install-pocketsphinx.md))

This is IMHO a reasonable distribution.

What I'm missing:
- [ ] There seems to be lines missing from the src to the dsts.
- [ ] I do not know, nor is it explained, why we use/need sphinxbase (could be explained in its new page)
- [ ] Checks like `gst-inspect-1.0 pocketsphinx` in the current state should go on pocketsphinx
- [ ] pocketsphinx is in fact gstreamer+pocketsphinx. Should be chopped or specified.

jgvictores, 2018-05-16:
Still some issues, tracking help here:
- https://github.com/roboticslab-uc3m/speech/pull/18
- https://github.com/roboticslab-uc3m/vision/issues/46

jgvictores, 2018-05-17:
Pretty happy with current state of:
- https://github.com/roboticslab-uc3m/speech/blob/41336b99e01fad4c8ad9244dd2c7fe18fc9182ee/doc/speech-install.md
- https://github.com/roboticslab-uc3m/installation-guides/blob/0820e41e13009904c70798219a59a5b38dbd4da4/install-gstreamer-pocketsphinx.md

Closing issue!

closed 2018-05-17 by jgvictores