roboticslab-uc3m/installation-guides #45: Document installation of FCL for ubuntu 14.04

jgvictores, 2018-05-08:
_From @rsantos88 on May 7, 2018 15:19_

Are [this](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-openrave.md) installation guide for Openrave updated?
Where is FCL installation for ubuntu 14.04 [here](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-openrave.md#install-openrave-with-fcl-confirmed-for-ubuntu-1504-1510-1610-1704)? 

_Copied from original issue: roboticslab-uc3m/teo-developer-manual#37_

jgvictores, 2018-05-08:
_From @PeterBowman on May 7, 2018 15:31_

1. Should be, at least more up-to-date than the ones I used long ago: [1](http://fsuarez6.github.io/blog/openrave-trusty/), [2](http://fsuarez6.github.io/blog/workstation-setup-xenial/).
2. Look at [*ref1, trusty, see next link if still in trouble with FCL*](https://github.com/roboticslab-uc3m/installation-guides/blob/9e10dce89fbe654b63b79d00c371b04401025ed7/install-openrave.md#external-installation-tutorialscript-links), which brings you back to the same website.

jgvictores, 2018-05-08:
> Are [this](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-openrave.md) installation guide for Openrave updated?

Yes, our latest version is definitely at `master` to which you link, 0fc5817181bf3fa6eef4d07b18f5a70091297da4 ATOW.

> Where is FCL installation for ubuntu 14.04 [here](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-openrave.md#install-openrave-with-fcl-confirmed-for-ubuntu-1504-1510-1610-1704)? 

Yes, FCL is in a separate section as it is not mandatory. If you confirm the guide works for 14.04 (iin fact, it should) you can add it to the section name. : )

jgvictores, 2018-05-08:
Moved this issue to https://github.com/roboticslab-uc3m/installation-guides, made more sense. : )

rsantos88, 2018-05-08:
> 2. Look at [*ref1, trusty, see next link if still in trouble with FCL*](https://github.com/roboticslab-uc3m/installation-guides/blob/9e10dce89fbe654b63b79d00c371b04401025ed7/install-openrave.md#external-installation-tutorialscript-links), which brings you back to the same website.

Yes, I've always used this scripts because for me it has never given me problems and it works great.
I just wanted to know if this guide was updated. It was to make sure. Thanks.

> Yes, FCL is in a separate section as it is not mandatory. If you confirm the guide works for 14.04 (iin fact, it should) you can add it to the section name. : )

When I try to install FCL in Ubuntu 14.04 like the guide, it can't be found:
```shell
teo@teo-oliver:~$ sudo apt install libfcl-dev
[sudo] password for teo: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libfcl-dev
```
You need to clone the repository and compile like the script.
Once clarified, I will close the issue.

closed 2018-05-08 by rsantos88

jgvictores, 2018-05-08:
Just for reference, our guide does **not** state the `libfcl-dev` package on 14.04 (Trusty).

In the following link you can see on which Ubuntu versions **do** contain the package: https://packages.ubuntu.com/search?keywords=libfcl-dev

rsantos88, 2018-05-08:
> If you confirm the guide works for 14.04 (iin fact, it should) you can add it to the section name. : )

Sorry, I thought you were referring to FCL

jgvictores, 2018-05-08:
Added comment at 8048966a29ea1f274e7d8b6fdf09a65284a34d81#diff-fc3039c132fc459d3ac5d0ad0464d573 (and fixed typos at next two commits).