roboticslab-uc3m/installation-guides #48: [yarp] updates for bionic

jgvictores, 2018-06-04:
```
E: Unable to locate package qtdeclarative5-qtmultimedia-plugin
E: Unable to locate package qtdeclarative5-controls-plugin
E: Unable to locate package qtdeclarative5-dialogs-plugin
```

jgvictores, 2018-06-04:
Fixes are good at http://www.yarp.it/install_yarp_linux.html#install_qt5_debian_jessie (or perhaps a mix):
```
sudo apt-get install qtbase5-dev qtdeclarative5-dev qtmultimedia5-dev \
  qml-module-qtquick2 qml-module-qtquick-window2 \
  qml-module-qtmultimedia qml-module-qtquick-dialogs \
  qml-module-qtquick-controls
```

davidvelascogarcia, 2018-07-10:
Tested on Lubuntu 18.04 LTS. Works very well.

imontesino, 2018-09-28:
Issues with yarpview:

```
$ yarpview
QQmlApplicationEngine failed to load component
qrc:/qml/QtYARPView/main.qml:60 Type VideoSurface unavailable
qrc:/YARPView/VideoSurface.qml:423 Type FileDialog unavailable
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:48 module "Qt.labs.settings" is not installed
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:47 module "Qt.labs.folderlistmodel" is not installed
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:48 module "Qt.labs.settings" is not installed
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:47 module "Qt.labs.folderlistmodel" is not installed
```

jgvictores, 2018-09-28:
From https://github.com/robotology/yarp/issues/1238:
- qml-module-qt-labs-folderlistmodel
- qml-module-qt-labs-settings

PeterBowman, 2018-11-18:
http://www.yarp.it/dependencies.html#dependencies_qt5_debian_ubuntu

RaulFdzbis, 2019-02-06:
worked for me too in ubuntu 18.04 bionic.

Thanks!

PeterBowman, 2019-12-14:
Now we have:

```
qtbase5-dev qtdeclarative5-dev qtmultimedia5-dev \
  qtdeclarative5-qtquick2-plugin qtdeclarative5-window-plugin \
  qtdeclarative5-qtmultimedia-plugin qtdeclarative5-dialogs-plugin \
  qtdeclarative5-controls-plugin libqt5svg5
```

Official install guides ([ref1](http://www.yarp.it/install_yarp_linux.html#install_linux_from_sources) and [ref2](http://www.yarp.it/dependencies.html#dependencies_qt5_debian_ubuntu)) list these instead:

```
qtbase5-dev qtdeclarative5-dev qtmultimedia5-dev \
  qml-module-qtquick2 qml-module-qtquick-window2 \
  qml-module-qtmultimedia qml-module-qtquick-dialogs \
  qml-module-qtquick-controls qml-module-qt-labs-folderlistmodel
  qml-module-qt-labs-settings
```

I've learned that `qtdeclarative5-<name>-plugin` is exactly the same as `qml-module-{qtquick}-<name>`, whereas the latter seems the modern way of packaging Qt5 stuff via apt. Since everything in the second list is available in Xenial, and the first list maps to the second one via meta-packages (not accounting for `qml-module-qt-labs-folderlistmodel qml-module-qt-labs-settings`, which nonetheless are also available in Xenial), I will update our install guides accordingly.

Note: Trusty is now a legacy OS, refer to YARP docs in order to pick the correct apt packages.

closed 2019-12-14 by PeterBowman