roboticslab-uc3m/installation-guides #49: Problemas instalando YCM.

serna92, 2018-06-24:
Buenas, estoy realizando un Trabajo Fin de Master relacionado con la simulación del robot Asibot en un entorno de cocina. Antes de la implementación con YCM me funcionaba pero ahora tengo problemas para adaptar lo que tenía hecho con el código nuevo. Ahora intentando instalar YCM 0.8+ me da error al usar el comando `make -j$(nproc)`:

> Scanning dependencies of target modules
> Scanning dependencies of target cmake-proposed
> Scanning dependencies of target find-modules
> Scanning dependencies of target cmake-f248f8ad7898816a5a627e288d3a04fed993967f
> [  2%] Downloading file Modules/SelectLibraryConfigurations.cmake from CMake git repository (ref f248f8ad7898816a5a627e288d3a04fed993967f)
> -- Cannot download file https://raw.githubusercontent.com/Kitware/CMake/f248f8ad7898816a5a627e288d3a04fed993967f/Modules/SelectLibraryConfigurations.cmake
>   Network problem or not existing file.
>   CMake Error at /home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/ycm_download_Modules_SelectLibraryConfigurations_cmake_real.cmake:9 (file):
>   file DOWNLOAD HASH mismatch
> 
>     for file: [/home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/downloads/Modules/SelectLibraryConfigurations.cmake]
>       expected hash: [e09bf3cd3f1faa1f9cc8f5924216a7fe3e4594d4]
>         actual hash: [da39a3ee5e6b4b0d3255bfef95601890afd80709]
>              status: [1;"Unsupported protocol"].
>   Retrying.
> 
> -- Cannot download file https://raw.githubusercontent.com/Kitware/CMake/f248f8ad7898816a5a627e288d3a04fed993967f/Modules/SelectLibraryConfigurations.cmake
>   Network problem or not existing file.
>   CMake Error at /home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/ycm_download_Modules_SelectLibraryConfigurations_cmake_real.cmake:9 (file):
>   file DOWNLOAD HASH mismatch
> 
>     for file: [/home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/downloads/Modules/SelectLibraryConfigurations.cmake]
>       expected hash: [e09bf3cd3f1faa1f9cc8f5924216a7fe3e4594d4]
>         actual hash: [da39a3ee5e6b4b0d3255bfef95601890afd80709]
>              status: [1;"Unsupported protocol"].
>   Retrying.
> 
> [  5%] Downloading file Copyright.txt from CMake git repository (ref f248f8ad7898816a5a627e288d3a04fed993967f)
> CMake Error at /home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/ycm_download_Modules_SelectLibraryConfigurations_cmake.cmake:30 (message):
>   Cannot download file
>   https://raw.githubusercontent.com/Kitware/CMake/f248f8ad7898816a5a627e288d3a04fed993967f/Modules/SelectLibraryConfigurations.cmake
> 
> 
>     Network problem or not existing file.
>     CMake Error at /home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/ycm_download_Modules_SelectLibraryConfigurations_cmake_real.cmake:9 (file):
>     file DOWNLOAD HASH mismatch
> 
>   
> 
>       for file: [/home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/downloads/Modules/SelectLibraryConfigurations.cmake]
>         expected hash: [e09bf3cd3f1faa1f9cc8f5924216a7fe3e4594d4]
>           actual hash: [da39a3ee5e6b4b0d3255bfef95601890afd80709]
>                status: [1;"Unsupported protocol"]
> 
> 
> cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/build.make:115: fallo en las instrucciones para el objetivo 'cmake-next/cmake-next/Modules/SelectLibraryConfigurations.cmake'
> make[2]: *** [cmake-next/cmake-next/Modules/SelectLibraryConfigurations.cmake] Error 1
> CMakeFiles/Makefile2:1157: fallo en las instrucciones para el objetivo 'cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/all'
> make[1]: *** [cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/all] Error 2
> make[1]: *** Se espera a que terminen otras tareas....
> -- Cannot download file https://raw.githubusercontent.com/Kitware/CMake/f248f8ad7898816a5a627e288d3a04fed993967f/Copyright.txt
>   Network problem or not existing file.
>   CMake Error at /home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/ycm_download_Copyright_txt_real.cmake:9 (file):
>   file DOWNLOAD HASH mismatch
> 
>     for file: [/home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/downloads/Copyright.txt]
>       expected hash: [96563e603328872bc22d31a3445e23b438292751]
>         actual hash: [da39a3ee5e6b4b0d3255bfef95601890afd80709]
>              status: [1;"Unsupported protocol"].
>   Retrying.
> 
> -- Cannot download file https://raw.githubusercontent.com/Kitware/CMake/f248f8ad7898816a5a627e288d3a04fed993967f/Copyright.txt
>   Network problem or not existing file.
>   CMake Error at /home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/ycm_download_Copyright_txt_real.cmake:9 (file):
>   file DOWNLOAD HASH mismatch
> 
>     for file: [/home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/downloads/Copyright.txt]
>       expected hash: [96563e603328872bc22d31a3445e23b438292751]
>         actual hash: [da39a3ee5e6b4b0d3255bfef95601890afd80709]
>              status: [1;"Unsupported protocol"].
>   Retrying.
> 
> CMake Error at /home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/ycm_download_Copyright_txt.cmake:30 (message):
>   Cannot download file
>   https://raw.githubusercontent.com/Kitware/CMake/f248f8ad7898816a5a627e288d3a04fed993967f/Copyright.txt
> 
> 
>     Network problem or not existing file.
>     CMake Error at /home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/ycm_download_Copyright_txt_real.cmake:9 (file):
>     file DOWNLOAD HASH mismatch
> 
>   
> 
>       for file: [/home/miguel/repos/ycm/build/cmake-next/CMakeFiles/cmake-f248f8ad7898816a5a627e288d3a04fed993967f.dir/downloads/Copyright.txt]
>         expected hash: [96563e603328872bc22d31a3445e23b438292751]
>           actual hash: [da39a3ee5e6b4b0d3255bfef95601890afd80709]
>                status: [1;"Unsupported protocol"]
> 
> 
> cmake-next/CMakeFiles/cmake-proposed.dir/build.make:98: fallo en las instrucciones para el objetivo 'cmake-next/cmake-next/Copyright.txt'
> make[2]: *** [cmake-next/cmake-next/Copyright.txt] Error 1
> CMakeFiles/Makefile2:1194: fallo en las instrucciones para el objetivo 'cmake-next/CMakeFiles/cmake-proposed.dir/all'
> [  7%] Files from modules installed in CMake build directory
> make[1]: *** [cmake-next/CMakeFiles/cmake-proposed.dir/all] Error 2
> [  7%] Built target modules
> [ 10%] Files from find-modules installed in CMake build directory
> [ 10%] Built target find-modules
> Makefile:128: fallo en las instrucciones para el objetivo 'all'
> make: *** [all] Error 2

Tengo bastante urgencia con este tema y agradecería cualquier documentación útil para realizar una simulación con el robot Asibot.

Muchas gracias por adelantado y disculpen las molestias.

serna92, 2018-06-24:
Tengo Ubuntu sudo 16.04.1 LTS (Xenial Xerus) y la versión de cmake es 3.11.0.

PeterBowman, 2018-06-24:
Los paquetes para Xenial solo llegan a 3.5, luego es muy probable que lo hayas instalado manualmente. Prueba activando las opciones que aparecen en aquella guía, y recomendaría usar también `-DBUILD_CursesDialog=ON` para generar `ccmake`.

PeterBowman, 2018-07-27:
Cerrando por inactividad. He mejorado un poco la sección de soporte para OpenSSL en nuestra guía de instalación de CMake (https://github.com/roboticslab-uc3m/installation-guides/commit/d40ea7498c0fb2457ff369c70c011848b2e67230) y la he referenciado desde la de YCM (https://github.com/roboticslab-uc3m/installation-guides/commit/cff0ca3807aebf06dfc4b9aa7388eb008ea982eb).

closed 2018-07-27 by PeterBowman