roboticslab-uc3m/installation-guides #50: [yarp] LibOVR (>=1.3) : not found

holygame, 2018-07-10:
What do i need to do to have it found ? i have the Oculus SDK and i've created an ENV variable Named OculusSDK_Root pointing to it's root. I'm using Visual Studio 2017.

PeterBowman, 2018-07-10:
The actual script that searches for LibOVR on your system is [here](https://github.com/robotology/yarp/blob/e2260fa5f4f19b71d536b33ba151fe7e6697f569/extern/ycm/ycm-0.8.20180628.11/find-modules/FindLibOVR.cmake). I'm a bit suspicious about the env variable you have set, make sure you are defining `OculusSDK_ROOT` (mind the capitalization). In case of further trouble, launch the CMake GUI and look for entries starting with `LibOVR_` (you'll need to mark the *Advanced* option) that resolve to `NOTFOUND`.

holygame, 2018-07-10:
Thanks a lot! Turning on the advanced option made me realise i had to build the LibOVRKernel.lib ( release and debug ) , once done it all worked perfectly!

closed 2018-07-10 by holygame