roboticslab-uc3m/installation-guides #51: [yarp] errors with ovrheadset device

holygame, 2018-07-10:
i get this list of errors when trying to build the ovrheadset device , i have the latest oculus SDK ( 1.26 ) and using the latest version of YARP 
```
Error	C3861	'getRate': identifier not found	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\TextureBattery.cpp	93	
Error	C3861	'ovr_SetInt32': identifier not found	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	778	
Error	C3861	'getEstPeriod': identifier not found	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	850	
Error	C3861	'getEstUsed': identifier not found	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	851	
Error	C2039	'HmdToEyeOffset': is not a member of 'ovrEyeRenderDesc_'	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	934	
Error	C2664	'void ovr_CalcEyePoses2(ovrPosef,const ovrPosef [],ovrPosef [])': cannot convert argument 2 from 'ovrVector3f [2]' to 'const ovrPosef []'	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	936	
Error	C2039	'*': is not a member of 'yarp::math'	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	963	
Error	C2039	'*': is not a member of 'yarp::math'	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	964	
Error	C3861	'ovr_GetInt32': identifier not found	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	1389	
Error	C3861	'ovr_SetInt32': identifier not found	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	1391	

```

PeterBowman, 2018-07-10:
There is an ongoing pull request that aims to fix this code under latest YARP (3.0), see https://github.com/robotology/yarp/pull/1789. As I see it, you can either downgrade your Oculus framework **and** downgrade YARP (because of the `yarp::math` errors), or clone that YARP fork provided it fully solves the compilation issues.

holygame, 2018-07-10:
the YARP fork provided solves all the issues but the errors:
```
Error	C2039	'HmdToEyeOffset': is not a member of 'ovrEyeRenderDesc_'	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	934	
Error	C2664	'void ovr_CalcEyePoses2(ovrPosef,const ovrPosef [],ovrPosef [])': cannot convert argument 2 from 'ovrVector3f [2]' to 'const ovrPosef []'	yarp_ovrheadset	C:\Users\user\Desktop\yarp\src\devices\ovrheadset\OVRHeadset.cpp	936	
```
to which Oculus framework should i downgrade to solve those?

PeterBowman, 2018-07-10:
Try with 1.16, it seems the last one to provide `HmdToEyeOffset`, and `ovr_GetEyePoses` is still there (recent releases shadow it with `ovr_GetEyePoses2`): [ref1](https://developer.oculus.com/reference/libovr/1.16/structovr_eye_render_desc/#ac4e5ddae21296f1e2815d1c6cd1c27c7), [ref2](https://developer.oculus.com/reference/libovr/1.16/o_v_r_c_a_p_i_util_8h/#ae1381f9c446511ee2e741198ec5c3f74), [ref3](https://developer.oculus.com/reference/libovr/1.26/o_v_r_c_a_p_i_util_8h/).

holygame, 2018-07-10:
Thank you , that solved the issue !

closed 2018-07-10 by holygame