roboticslab-uc3m/installation-guides #52: [YARP] Errors with ovrheadset device library

holygame, 2018-07-11:
I have some linker issues when using the ovrheadset device , It is might be because it's not recognized  by the command ` yarpdev --list` . I get the following error message when using the command

> error while opening C:/Program Files (x86)/YARP/lib/yarp/yarp_ovrheadset.dll:
>   Le module spÚcifiÚ est introuvable. 
> 
> Le module spÚcifiÚ est introuvable.
> 
> error while opening C:/Program Files (x86)/YARP/lib/yarp/yarp_ovrheadsetd.dll:
>   Le module spÚcifiÚ est introuvable.
> 
> Le module spÚcifiÚ est introuvable.

and a warning message

>  [WARNING]Wrong library name for plugin ovrheadset

the other devices works just fine , my ovrheadset.ini specify the correct library name (yarp_ovrheadset.dll) and the library does exists at the correct path so i really can't figure out ,what could be the cause of this problem ? I'm using the last version of Yarp and the Oculus SDK v1.16.

PeterBowman, 2018-07-11:
Perhaps some DLLs are missing from the `PATH` env variable, that is, `yarp_ovrheadset.dll` depends on any other `.dll` that casually cannot be found on runtime. You can list these dependencies with http://www.dependencywalker.com, just drop `yarp_ovrheadset.dll` into the window and look for library names listed with a question mark (those will mostly show up in the first/uppermost level of the dependency tree). Dismiss the usual `API-MS-WIN-CRT...` DLLs, and also check for architecture mismatches (e.g. 32-bit DLL but 64-bit dependency).

holygame, 2018-07-12:
Thank you so much for the help . I was missing Glew32 and GLFW3 DLLs from the PATH env. I think this bug occured since i manually defined the path for those libraries using CMAKE GUI as the FindGlew and FindGLFW3.cmake doesn't find them correctly.

PeterBowman, 2018-07-12:
Glad to help :). Actually, any path defined through CMake is only relevant during the build phase (i.e. compilation + linking). It does not affect how your program behaves on runtime, that is, once you launch it. For that to work properly, you need either to tweak your PATH variable, or copy any DLL listed as a dependency by DependencyWalker and paste it in the same directory as your .exe (or in the current working directory when running everything from the command line). With YARP devices loaded dynamically, which requires more care, I totally recommend the former.

closed 2018-07-18 by holygame