roboticslab-uc3m/installation-guides #53: [openrave] tracking boost

jgvictores, 2018-07-13:
On Bionic, installing `libboost-dev` leads to:
```
-- Could NOT find Boost
-- found boost version: 106501
CMake Error at CMakeLists.txt:357 (message):
  boost thread and date_time libraries are necessary
```

Using Openrave:
```
commit 7c5f5e27eec2b2ef10aa63fbc519a998c276f908 (HEAD -> master, origin/master, origin/HEAD)
Merge: b2818fb5 7ed9c2be
Author: Rosen Diankov <rosen.diankov@gmail.com>
Date:   Mon Feb 19 11:25:45 2018 +0900

    Merge pull request #557 from rdiankov/verification
    
    Verification
```

jgvictores, 2018-07-13:
`sudo apt install libboost-thread-dev libboost-date-time-dev` leads to (where `--no-install-recommends` has no effect):
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libboost-atomic1.65-dev libboost-atomic1.65.1 libboost-chrono1.65-dev libboost-chrono1.65.1
  libboost-date-time1.65-dev libboost-serialization1.65-dev libboost-serialization1.65.1
  libboost-system1.65-dev libboost-thread1.65-dev
The following NEW packages will be installed:
  libboost-atomic1.65-dev libboost-atomic1.65.1 libboost-chrono1.65-dev libboost-chrono1.65.1
  libboost-date-time-dev libboost-date-time1.65-dev libboost-serialization1.65-dev
  libboost-serialization1.65.1 libboost-system1.65-dev libboost-thread-dev libboost-thread1.65-dev
```

jgvictores, 2018-07-13:
```
/usr/bin/x86_64-linux-gnu-ld: CMakeFiles/openrave.dir/openrave.cpp.o: undefined reference to symbol '_ZN5boost6system15system_categoryEv'
//usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1: error adding symbols: DSO missing from command line
```
:scream:

PeterBowman, 2018-11-18:
> On Bionic, installing `libboost-dev` leads to:

This was expected, `libboost-dev` is a mere container for all Boost **header files**. In case a Boost non-header-only component is required (such as Boost.Thread), you'll need to install the corresponding package with said component's **binaries**. Incidentally, any `libboost-<component>-dev` package depends on `libboost-dev`. I guess you wanted `libboost-all-dev`, which actually installs all Boost stuff at once.

See https://github.com/roboticslab-uc3m/installation-guides/commit/46d51600aacc5175463bb208f55e071193628712 and https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/26c4edd66e3be8947c7a2a06c80f2a5ff54bbc47.

jgvictores, 2018-11-18:
Sorry for not specifying this clearly in the issue description: the aim was towards being more minimalistic. Why install all Boost non-header, when we can install only on certain components?

Note: The answer may be "Boost components have too many mutual dependencies, this makes no sense".

PeterBowman, 2018-11-18:
> Why install all Boost non-header, when we can install only on certain components?

I fully agree with you, see my approach in the linked commit at the openrave-yarp-plugins repo. Sadly, it's hard to determine which Boost components are required by OpenRAVE, and even those could change without prior notice (unless rdiankov keeps an up-to-date install guide somewhere).

PeterBowman, 2019-02-21:
> error adding symbols: DSO missing from command line

Run into this today, was using OR 0.9 on Xenial. Make sure you pass all required components to `find_package()`, e.g.:

```cmake
find_package(Boost "${OpenRAVE_Boost_VERSION}" REQUIRED
             COMPONENTS system)
```

PeterBowman, 2019-11-21:
Issue https://github.com/roboticslab-uc3m/installation-guides/issues/65 provides two solutions:

* downgrade Boost (1.65 > 1.58), see https://github.com/crigroup/openrave-installation
* either wait until https://github.com/rdiankov/openrave/pull/625 is merged or use https://github.com/cielavenir/openrave/tree/boost-1.6x-forcompile

Regarding the original intent:

> Why install all Boost non-header, when we can install only on certain components?

OK, I have updated the title accordingly. I'd say WONTFIX to this since everything related to external dependencies and CMake in OpenRAVE seems totally unreliable to me. But I'd take a final look in case it's not that hard after all.

PeterBowman, 2019-12-14:
> Sadly, it's hard to determine which Boost components are required by OpenRAVE, and even those could change without prior notice (unless rdiankov keeps an up-to-date install guide somewhere).

There it is, unchanged for nine years ([ref](https://github.com/rdiankov/openrave/blob/abe85928e74381d3d3d33db73693d92c9fbd7a1c/CMakeLists.txt#L315)):

```cmake
find_package(Boost COMPONENTS regex filesystem system python thread iostreams date_time)
```

[This SO answer](https://stackoverflow.com/a/13604163) lists Boost libraries that require building ([official link](https://www.boost.org/doc/libs/1_72_0/more/getting_started/unix-variants.html#header-only-libraries) regarding Unix systems). Ubuntu packages are listed [here](https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=libboost&searchon=names) (limited to 100 results). So, we need to install these packages as Boost dependencies for OpenRAVE given the above list:

* libboost-filesystem-dev
* libboost-system-dev
* libboost-python-dev
* libboost-thread-dev
* libboost-iostreams-dev

Next, I'm going to update the OR installation guide and our Travis build scripts

closed 2019-12-14 by PeterBowman