roboticslab-uc3m/installation-guides #54: [YARP] can't link with ovrheadset lib on windows (MSVC 17)

holygame, 2018-07-18:
I have an issue building a project using the ovrheadset library , i'm linking the correct library "yarp_ovrheadset.lib" but still get this errors. 

```
1>OvrTest.obj : error LNK2019: unresolved external symbol "public: __cdecl yarp::dev::OVRHeadset::OVRHeadset(void)" (??0OVRHeadset@dev@yarp@@QEAA@XZ) referenced in function main
1>OvrTest.obj : error LNK2019: unresolved external symbol "public: virtual __cdecl yarp::dev::OVRHeadset::~OVRHeadset(void)" (??1OVRHeadset@dev@yarp@@UEAA@XZ) referenced in function "public: virtual __cdecl FakeOvr::~FakeOvr(void)" (??1FakeOvr@@UEAA@XZ)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::threadInit(void)" (?threadInit@OVRHeadset@dev@yarp@@UEAA_NXZ)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual void __cdecl yarp::dev::OVRHeadset::threadRelease(void)" (?threadRelease@OVRHeadset@dev@yarp@@UEAAXXZ)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual void __cdecl yarp::dev::OVRHeadset::run(void)" (?run@OVRHeadset@dev@yarp@@UEAAXXZ)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::startService(void)" (?startService@OVRHeadset@dev@yarp@@UEAA_NXZ)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::updateService(void)" (?updateService@OVRHeadset@dev@yarp@@UEAA_NXZ)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::stopService(void)" (?stopService@OVRHeadset@dev@yarp@@UEAA_NXZ)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getAxisCount(unsigned int &)" (?getAxisCount@OVRHeadset@dev@yarp@@UEAA_NAEAI@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getButtonCount(unsigned int &)" (?getButtonCount@OVRHeadset@dev@yarp@@UEAA_NAEAI@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getTrackballCount(unsigned int &)" (?getTrackballCount@OVRHeadset@dev@yarp@@UEAA_NAEAI@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getHatCount(unsigned int &)" (?getHatCount@OVRHeadset@dev@yarp@@UEAA_NAEAI@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getTouchSurfaceCount(unsigned int &)" (?getTouchSurfaceCount@OVRHeadset@dev@yarp@@UEAA_NAEAI@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getStickCount(unsigned int &)" (?getStickCount@OVRHeadset@dev@yarp@@UEAA_NAEAI@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getStickDoF(unsigned int,unsigned int &)" (?getStickDoF@OVRHeadset@dev@yarp@@UEAA_NIAEAI@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getButton(unsigned int,float &)" (?getButton@OVRHeadset@dev@yarp@@UEAA_NIAEAM@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getTrackball(unsigned int,class yarp::sig::Vector &)" (?getTrackball@OVRHeadset@dev@yarp@@UEAA_NIAEAVVector@sig@3@@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getHat(unsigned int,unsigned char &)" (?getHat@OVRHeadset@dev@yarp@@UEAA_NIAEAE@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getAxis(unsigned int,double &)" (?getAxis@OVRHeadset@dev@yarp@@UEAA_NIAEAN@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getStick(unsigned int,class yarp::sig::Vector &,enum yarp::dev::IJoypadController::JoypadCtrl_coordinateMode)" (?getStick@OVRHeadset@dev@yarp@@UEAA_NIAEAVVector@sig@3@W4JoypadCtrl_coordinateMode@IJoypadController@23@@Z)
1>OvrTest.obj : error LNK2001: unresolved external symbol "public: virtual bool __cdecl yarp::dev::OVRHeadset::getTouch(unsigned int,class yarp::sig::Vector &)" (?getTouch@OVRHeadset@dev@yarp@@UEAA_NIAEAVVector@sig@3@@Z)
1>C:\Users\AnDy\Yarp_Projects\OvrDeviceTests\build\Release\OvrTest.exe : fatal error LNK1120: 21 unresolved externals
1>Done building project "OvrTest.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 1 up-to-date, 0 skipped ==========
```

here is the CMakeLists.txt 
```

# YARP needs CMake 3.0 or greater
cmake_minimum_required(VERSION 3.5)

# find YARP
find_package(YARP REQUIRED)

# set up our program
add_executable(OvrTest OvrTest.cpp)

include_directories("C:/OculusSDK_V1.16/OculusSDK/LibOVR/Include" 
	"C:/OculusSDK_V1.16/OculusSDK/LibOVR/Include/Extras"
	"C:/OculusSDK_V1.16/OculusSDK/LibOVRKernel/Src"
	"C:/OculusSDK_V1.16/OculusSDK/LibOVRKernel/Src/Kernel"
	"C:/Users/AnDy/source/glew-2.1.0/include"
	"C:/Program\ Files/GLFW/include"
	"C:/Program\ Files\ (x86)/yarp-2.3.72/src/devices/ovrheadset"
	"C:/Program\ Files\ (x86)/yarp-2.3.72/build/src/devices/ovrheadset")

# link with YARP libraries
target_link_libraries(OvrTest ${YARP_LIBRARIES} 
			"C:/Program\ Files/YARP/lib/yarp/yarp_ovrheadset.dll")

message("yarp libs : " ${YARP_LIBRARIES})

```
I've installed yarp in C:\Program Files\YARP and I have the builds in C:\Program Files (x86)\yarp-2.3.72 ( I installed the 64bit version )

PeterBowman, 2018-07-18:
```cmake
target_link_libraries(OvrTest ${YARP_LIBRARIES} 
			"C:/Program\ Files/YARP/lib/yarp/yarp_ovrheadset.dll")
```

* In a general case, you should never link against dynamic libraries (.dll), use their static counterpart instead (.lib).
* YARP devices (such as OVRHeadset) can be compiled as dynamically loaded libraries a.k.a. plugins. You interact with them on **runtime** through a layer of indirection (a C++ interface), hence your code should probably be updated. Please see http://www.yarp.it/note_devices.html.

Also, beware of architecture mismatches. Linking a 32-bit library/application with 64-bit binaries (or viceversa) will lead to undesired results.

PeterBowman, 2018-07-18:
The errors are actually expected. Without seeing your code I can only guess that `yarp::dev::OVRHeadset` is instantiated at some point (or derived from), which is a misconception of how YARP devices work. Actually, you are setting the include path to the source directory of the device (nota bene it relates to a 32-bit build), so that's why the code even compiled in the first place (and failed to link thereafter).

Take a careful look at the [*Creating and configuring the new device* section](http://www.yarp.it/note_devices.html#note_devices_library). Roughly, all you need is:

```c++
yarp::dev::PolyDriver dd;

if (!dd.open("ovrheadset"))
{
    // error
}

yarp::dev::IJoypadController * iJoyPad;
dd.view(iJoyPad);

// optionally, check that iJoyPad is not 0/NULL/nullptr

// now you can access the ovrheadset device through a
// handle (iJoyPad) to the IJoypadController interface
```

holygame, 2018-07-18:
wow...
i've been stuck on this for so long! It's more intuitive to just derive from the yarp::dev::OVRHeadset class , i'd never tought it doesn't work like that. Thanks a lot !

holygame, 2018-07-19:
@PeterBowman Excuse me to ask such basic questions but when I used your code to open the driver,
 I got a list of parameters that needs to be initialized, however I don't quit get their meaning to do so. 
Could you please help me find documentations or explain it to me? Mainly "tfDevice" and "tfRemote".

closed 2018-07-25 by holygame