roboticslab-uc3m/installation-guides #55: Problem using Aquila

waterzch, 2018-11-17:


PeterBowman, 2018-11-17:
You can't just link a static library (`libAquila.a`) into a shared object unless it's specifically prepared for that use case, see [this SO question](https://stackoverflow.com/q/19364969/10404307). To not meddle with the source code of Aquila, instruct CMake to build a shared library instead: `cmake .. -DBUILD_SHARED_LIBS=ON`. You should see a `libAquila.so` in your build folder.

If that fails, add the following after [these lines](https://github.com/zsiciarz/aquila/blob/d5e3bde3c8d07678a95f225b657a7bb23d0c4730/CMakeLists.txt#L184-L185):

```cmake
set_target_properties(Aquila PROPERTIES POSITION_INDEPENDENT_CODE ON)
```

PeterBowman, 2018-12-19:
> To not meddle with the source code of Aquila, instruct CMake to build a shared library instead: `cmake .. -DBUILD_SHARED_LIBS=ON`.

Successfully tested on a custom shared library that links to `libAquila.so`. Closing with https://github.com/roboticslab-uc3m/installation-guides/commit/dd2938dbe0b4b05240945062afcb04fe208fead5.

closed 2018-12-19 by PeterBowman