roboticslab-uc3m/installation-guides #56: [openrave] Logging plugin with FFMPEG enabled and undefined reference

PeterBowman, 2018-12-30:
[‘avformat_alloc_output_context2’ was not declared in this scope](https://travis-ci.org/roboticslab-uc3m/teo-main/jobs/473466823#L8182):

```
/home/travis/build/roboticslab-uc3m/teo-main/external/OpenRAVE/plugins/logging/viewerrecorder.cpp: In member function ‘void ViewerRecorder::_StartVideo(const string&, double, int, int, int, int)’:
/home/travis/build/roboticslab-uc3m/teo-main/external/OpenRAVE/plugins/logging/viewerrecorder.cpp:735:71: error: ‘avformat_alloc_output_context2’ was not declared in this scope
             avformat_alloc_output_context2(&_output, NULL, "mp4", NULL);
                                                                       ^
```

Does not happen in asibot-main, turns out

jgvictores, 2018-12-30:
See comment in https://github.com/roboticslab-uc3m/installation-guides/blob/dd2938dbe0b4b05240945062afcb04fe208fead5/install-openrave.md:
```bash
cmake ..  # Use -DOPT_VIDEORECORDING=OFF if there are AV errors
```
As far as I recall, `AV`-prefixes are related to FFmpeg (example: https://trac.ffmpeg.org/wiki/Encode/AV1), which from old comments in YARP was said to constantly be changing API and hard to keep up with. Just to continue with YARP references, here's a recent issue/PR: https://github.com/robotology/yarp/issues/1849

PeterBowman, 2018-12-30:
No error at [openrave-yarp-plugins](https://travis-ci.org/roboticslab-uc3m/openrave-yarp-plugins/builds/471979215), and no clue (yet) which dependency triggers this behavior. Need to review https://github.com/rdiankov/openrave/pull/546 and probably add some header guard matching the macro definitions set in [CMakeLists.txt](https://github.com/rdiankov/openrave/blob/737c4b1bbeb4d32f8aef1bf3a027eedb93854c26/plugins/logging/CMakeLists.txt).

PeterBowman, 2019-01-19:
BTW we support OpenRAVE 0.15 (latest master), now. Repo-specific installation guides keep the 0.9+ requirement, and `find_package()` calls should not request a specific release number due to a strict upstream version check. That is, `find_package(OpenRAVE 0.9)` will **not** accept OR 0.15 if found on system. See https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/e0c92463bcaa56638273c658d58bbf69cb97514e.