roboticslab-uc3m/installation-guides #65: [openrave] install boost error with bionic

RaulFdzbis, 2019-02-06:
Installing Openrave [master] in Ubuntu 18.04 bionic gives this error:

```bash
[ 53%] Building CXX object python/bindings/CMakeFiles/convexdecompositionpy.dir/convexdecompositionpy.cpp.o
In file included from /usr/include/python2.7/numpy/ndarraytypes.h:1809:0,
                 from /usr/include/python2.7/numpy/ndarrayobject.h:18,
                 from /usr/include/python2.7/numpy/arrayobject.h:4,
                 from /home/raul/repos/openrave/python/bindings/convexdecompositionpy.cpp:20:
/usr/include/python2.7/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
In file included from /usr/include/python2.7/numpy/ndarraytypes.h:1809:0,
                 from /usr/include/python2.7/numpy/ndarrayobject.h:18,
                 from /usr/include/python2.7/numpy/arrayobject.h:4,
                 from /home/raul/repos/openrave/python/bindings/pyann.cpp:20:
/usr/include/python2.7/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
In file included from /home/raul/repos/openrave/python/bindings/convexdecompositionpy.cpp:29:0:
/home/raul/repos/openrave/python/bindings/bindings.h:357:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const float* pvalues, size_t N)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:370:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const float* pvalues, std::vector<npy_intp>& dims)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:389:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const double* pvalues, size_t N)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:402:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const double* pvalues, std::vector<npy_intp>& dims)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:421:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const uint8_t* pvalues, std::vector<npy_intp>& dims)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:440:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const uint8_t* pvalues, size_t N)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:453:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const int* pvalues, size_t N)
        ^~~~~~~
        cuserid
In file included from /home/raul/repos/openrave/python/bindings/pyann.cpp:29:0:
/home/raul/repos/openrave/python/bindings/bindings.h:357:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const float* pvalues, size_t N)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:466:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const uint32_t* pvalues, size_t N)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:370:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const float* pvalues, std::vector<npy_intp>& dims)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:490:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArray(const std::vector<T>& v)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:389:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const double* pvalues, size_t N)
        ^~~~~~~
        cuserid
/home/raul/repos/openrave/python/bindings/bindings.h:402:8: error: ‘numeric’ does not name a type; did you mean ‘cuserid’?
 inline numeric::array toPyArrayN(const double* pvalues, std::vector<npy_intp>& dims)
[... 218 lines cut ...]
/home/raul/repos/openrave/python/bindings/convexdecompositionpy.cpp:104:118: error: expected ‘>’ before ‘::’ token
         hulls.append(boost::python::make_tuple(static_cast<numeric::array>(handle<>(pyvertices)), static_cast<numeric::array>(handle<>(pyindices))));
                                                                                                                      ^~
/home/raul/repos/openrave/python/bindings/convexdecompositionpy.cpp:104:118: error: expected ‘(’ before ‘::’ token
/home/raul/repos/openrave/python/bindings/convexdecompositionpy.cpp:104:125: error: missing template arguments before ‘>’ token
         hulls.append(boost::python::make_tuple(static_cast<numeric::array>(handle<>(pyvertices)), static_cast<numeric::array>(handle<>(pyindices))));
                                                                                                                             ^
/home/raul/repos/openrave/python/bindings/convexdecompositionpy.cpp: In function ‘void init_module_convexdecompositionpy()’:
/home/raul/repos/openrave/python/bindings/convexdecompositionpy.cpp:115:5: error: ‘numeric’ has not been declared
     numeric::array::set_module_and_type("numpy", "ndarray");
     ^~~~~~~
python/bindings/CMakeFiles/pyANN_int.dir/build.make:62: recipe for target 'python/bindings/CMakeFiles/pyANN_int.dir/pyann.cpp.o' failed
make[2]: *** [python/bindings/CMakeFiles/pyANN_int.dir/pyann.cpp.o] Error 1
CMakeFiles/Makefile2:1191: recipe for target 'python/bindings/CMakeFiles/pyANN_int.dir/all' failed
make[1]: *** [python/bindings/CMakeFiles/pyANN_int.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
python/bindings/CMakeFiles/convexdecompositionpy.dir/build.make:62: recipe for target 'python/bindings/CMakeFiles/convexdecompositionpy.dir/convexdecompositionpy.cpp.o' failed
make[2]: *** [python/bindings/CMakeFiles/convexdecompositionpy.dir/convexdecompositionpy.cpp.o] Error 1
CMakeFiles/Makefile2:1229: recipe for target 'python/bindings/CMakeFiles/convexdecompositionpy.dir/all' failed
make[1]: *** [python/bindings/CMakeFiles/convexdecompositionpy.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

Possible related issues: https://github.com/rdiankov/openrave/issues/542 and https://github.com/rdiankov/openrave/issues/551

RaulFdzbis, 2019-02-06:
```
$ pip list | grep numpy
numpy                              1.15.4

RaulFdzbis, 2019-02-06:
```bash
$ sudo -H pip install --upgrade numpy
$ pip list | grep numpy
numpy                              1.16.1   
```

David-Estevez, 2019-02-07:
What is your Python Boost version?

RaulFdzbis, 2019-02-11:
Here you go:
```bash
~$ dpkg -s libboost-dev | grep 'Version'
Version: 1.65.1.0ubuntu1
```

David-Estevez, 2019-02-11:
Well, then it's definitely https://github.com/rdiankov/openrave/issues/551 :

> Building with Boost.Python version 1.65.1 or greater is broken, as the `boost::python::numeric` namespace has been removed in favor of `boost::python::numpy`

Options:
* Downgrade boost (in that issue they say that 1.58 is ok)
* Wait until they fix boost (use `upstream` tag)
* Accept your human fate and cry
[image]

RaulFdzbis, 2019-02-11:
Yeah it looks like it is. I always tend to choose the cry option, but i will keep you up if there is some news hahahaha. 

Thank you @David-Estevez !!

jgvictores, 2019-02-18:
Added `priority: high` as Bionic is growing in users, and it's not easy to simply uninstall boost for downgrade (many packages depend on it).

luckychay, 2019-10-10:
I met the same issue and I chose to cry for my fate.

PeterBowman, 2019-10-21:
OpenRAVE has been recently upgraded (from 0.15.0 to 0.24.1). As usual: some issues might be fixed now, some bugs may arise. For now, I identified a missing rapidjson dependency ([Travis build](https://travis-ci.com/roboticslab-uc3m/openrave-yarp-plugins/builds/132794360?utm_medium=notification&utm_source=email)).

PeterBowman, 2019-11-04:
> OpenRAVE has been recently upgraded (from 0.15.0 to 0.24.1). As usual: some issues might be fixed now, some bugs may arise.

The rapidjson regression is tracked at https://github.com/rdiankov/openrave/issues/707. Installing `rapidjson-dev` is not a complete fix since the `rapidjson/pointer.h` header file cannot be found afterwards: [ref](https://travis-ci.com/roboticslab-uc3m/openrave-yarp-plugins/jobs/252650010#L4295). Besides, a C++11 issue spoils gcc Travis jobs: [ref](https://travis-ci.com/roboticslab-uc3m/openrave-yarp-plugins/jobs/252650009#L1910).

cielavenir, 2019-11-04:
That travis says gcc 5.4.

gcc 6 (or later) defaults to C++14 and does not have emplace_back issue. Otherwise `-DCMAKE_CXX_FLAGS=-std=gnu++11` needs to be passed as described in https://github.com/cielavenir/mujin_recruiting/blob/master/cookbooks/mujin/recipes/default.rb#L132 (that default.rb describes openrave installation in devops way)

cielavenir, 2019-11-04:
For boost, you could work on https://github.com/rdiankov/openrave/pull/625 (note that I have not tested that one)

cielavenir, 2019-11-06:
I was able to establish the procedure to compile openrave on bionic (and buster) https://github.com/cielavenir/mujin_recruiting/blob/master/cookbooks/buster/recipes/default.rb

PeterBowman, 2019-11-06:
Thanks! I managed to compile it on bionic, too, following https://github.com/crigroup/openrave-installation. It does seem to work on latest OpenRAVE master (although the scripts hosted in there still point at v0.9) at the cost of forcing me to download and compile an older Boost release (if I'm correct, your code uses `apt`, which is nicer).

cielavenir, 2019-11-07:
@PeterBowman instead of using older boost, my installation procedure uses openrave specific branch supporting latest boost.

PeterBowman, 2019-11-08:
> Installing `rapidjson-dev` is not a complete fix since the `rapidjson/pointer.h` header file cannot be found afterwards: [ref](https://travis-ci.com/roboticslab-uc3m/openrave-yarp-plugins/jobs/252650010#L4295).

Probably fixed in Bionic since [rapidjson v1.1.0](https://packages.ubuntu.com/search?keywords=rapidjson-dev) already includes the offending *pointer.h* header ([ref](https://github.com/Tencent/rapidjson/tree/v1.1.0/include/rapidjson) at https://github.com/Tencent/rapidjson).

> Besides, a C++11 issue spoils gcc Travis jobs: [ref](https://travis-ci.com/roboticslab-uc3m/openrave-yarp-plugins/jobs/252650009#L1910).

> gcc 6 (or later) defaults to C++14 and does not have emplace_back issue.

See https://docs.travis-ci.com/user/languages/c/#gcc-on-linux for a guide on upgrading gcc in Travis. However, I'm hesitant about meddling with the build environment, probably more prone to change the CLI CMake options as mentioned in:

> Otherwise `-DCMAKE_CXX_FLAGS=-std=gnu++11` needs to be passed (...)

See https://stackoverflow.com/a/44357387.

RaulFdzbis, 2019-11-13:
Yeah! Being able to install it following https://github.com/crigroup/openrave-installation as stated by @PeterBowman. I did not try @cielavenir solution but looks good too. 

Thank you guys!

RaulFdzbis, 2019-11-13:
If you agree, will keep this issue open until there is a solid solution or at least is updated in the installation guides :D

RaulFdzbis, 2019-11-13:
**Update:**
After this 
> Yeah! Being able to install it following https://github.com/crigroup/openrave-installation ...

Getting this error while trying to run openrave:

```bash
2019-11-13 17:43:16,838 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /home/raul/repos/forceSensor/lib//libForceSensor.so: /home/raul/repos/forceSensor/lib//libForceSensor.so: undefined symbol: _ZNK8OpenRAVE13InterfaceBase20GetReadableInterfaceERKSs
2019-11-13 17:43:16,839 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /home/raul/repos/forceSensor/lib//libForceSensor.so: /home/raul/repos/forceSensor/lib//libForceSensor.so: undefined symbol: _ZNK8OpenRAVE13InterfaceBase20GetReadableInterfaceERKSs
2019-11-13 17:43:16,839 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /home/raul/repos/forceSensor/lib//libForceSensor.so: /home/raul/repos/forceSensor/lib//libForceSensor.so: undefined symbol: _ZNK8OpenRAVE13InterfaceBase20GetReadableInterfaceERKSs
2019-11-13 17:43:16,839 openrave [WARN] [plugindatabase.h:941 RaveDatabase::PluginPtr OpenRAVE::RaveDatabase::_LoadPlugin] failed to load: /home/raul/repos/forceSensor/lib//libForceSensor.so
2019-11-13 17:43:17,546 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so: libACE-6.0.3.so: cannot open shared object file: No such file or directory
2019-11-13 17:43:17,547 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so: libACE-6.0.3.so: cannot open shared object file: No such file or directory
2019-11-13 17:43:17,547 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so: libACE-6.0.3.so: cannot open shared object file: No such file or directory
2019-11-13 17:43:17,548 openrave [WARN] [plugindatabase.h:941 RaveDatabase::PluginPtr OpenRAVE::RaveDatabase::_LoadPlugin] failed to load: /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so
2019-11-13 17:43:18,778 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libqtosgrave.so: /usr/local/lib/openrave0.9-plugins/libqtosgrave.so: undefined symbol: _ZN3osg5Group11removeChildEPNS_4NodeE
Segmentation fault (core dumped)
```

Any idea?

PeterBowman, 2019-11-13:
Did you perform a fresh OpenRAVE installation? If not, perhaps some older symlinks are still pointing at the previous version. Check `ls -l /usr/local/lib/libopenrave0.*`.

jgvictores, 2019-11-13:
I'd also recompile the plugins:
- https://github.com/roboticslab-uc3m/openrave-yarp-plugins
-  https://github.com/robEllenberg/forceSensor (in your `/home/raul/repos/forceSensor`, possibly found via the path defined by your `OPENRAVE_PLUGINS` environment variable)

RaulFdzbis, 2019-11-14:
Did clean install of three of the packages (openrave, openraveyarp-plugins and forcesensors), solved problem with forceSensor library but still receiving segmentation fault:

```bash
2019-11-14 12:22:13,966 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so: libACE-6.0.3.so: cannot open shared object file: No such file or directory
2019-11-14 12:22:13,967 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so: libACE-6.0.3.so: cannot open shared object file: No such file or directory
2019-11-14 12:22:13,968 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so: libACE-6.0.3.so: cannot open shared object file: No such file or directory
2019-11-14 12:22:13,968 openrave [WARN] [plugindatabase.h:941 RaveDatabase::PluginPtr OpenRAVE::RaveDatabase::_LoadPlugin] failed to load: /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so
2019-11-14 12:22:14,183 openrave [WARN] [plugindatabase.h:1014 RaveDatabase::_SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libqtosgrave.so: /usr/local/lib/openrave0.9-plugins/libqtosgrave.so: undefined symbol: _ZN3osg5Group11removeChildEPNS_4NodeE
Segmentation fault (core dumped)
```

Looks like the problem comes with /usr/local/lib/openrave0.9-plugins/ not being correctly installed?

jgvictores, 2019-11-14:
Yes, please be sure to really completely wipe out old stuff and try again:
```bash
sudo rm -rf /usr/local/lib/openrave0.9-plugins
sudo rm -rf /home/raul/repos/openrave-yarp-plugins/build
cd /home/raul/repos/openrave-yarp-plugins
mkdir build; cd build; cmake ..
make
sudo make install
```

RaulFdzbis, 2019-11-14:
> Yes, please be sure to really completely wipe out old stuff and try again:
>```bash
>sudo rm -rf /usr/local/lib/openrave0.9-plugins
>sudo rm -rf /home/raul/repos/openrave-yarp-plugins/build
>cd /home/raul/repos/openrave-yarp-plugins
>mkdir build; cd build; cmake ..
>make
>sudo make install
>```

That solved the segmentation fault problem :D, however after that a new problem arised when running [openraveYarpPaintSquares.py](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/develop/examples/python/openraveYarpPaintSquares.py) D:

```bash
Rave has been initialized
2019-11-14 13:43:21,903 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name fcl_, interface collisionchecker
2019-11-14 13:43:21,903 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name ode, interface collisionchecker
2019-11-14 13:43:21,903 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name bullet, interface collisionchecker
2019-11-14 13:43:21,903 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name pqp, interface collisionchecker
2019-11-14 13:43:21,909 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name qtcoin, interface viewer
2019-11-14 13:43:21,910 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name ivmodelloader, interface module

DATA_READ - Fatal error!
  Unacceptable input file type.
2019-11-14 13:43:21,911 openrave [WARN] [xmlreaders-core.cpp:992 LinkXMLReader::endElement] failed to find /usr/local/share/teo-openrave-models/contexts/openrave/teo/models/RootWaist_links.wrl
2019-11-14 13:43:21,911 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name ivmodelloader, interface module

DATA_READ - Fatal error!
  Unacceptable input file type.
2019-11-14 13:43:21,914 openrave [WARN] [xmlreaders-core.cpp:1001 LinkXMLReader::endElement] failed to find /usr/local/share/teo-openrave-models/contexts/openrave/teo/models/RootWaist_links.wrl
2019-11-14 13:43:21,915 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name ivmodelloader, interface module
... 
```
**(output cut to avoid flooding the issue, same error for each part of the model)**

Running openrave brings the following output:

```bash
~$ openrave
2019-11-14 16:21:43,864 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name fcl_, interface collisionchecker
2019-11-14 16:21:43,864 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name ode, interface collisionchecker
2019-11-14 16:21:43,864 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name bullet, interface collisionchecker
2019-11-14 16:21:43,864 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name pqp, interface collisionchecker
2019-11-14 16:21:43,864 openrave [WARN] [openrave.cpp:322 MainOpenRAVEThread] failed to find an OpenRAVE viewer.
```

Thanks to @jgvictores we have come to the conclusion that the load of the *.wrl* models can depend on the viewer used by Openrave, so the error may come from not having installed the viewer in Openrave. I will try to change  https://github.com/crigroup/openrave-installation configuration to force the installation of the qtcoin viewer and see if this can fix the problem.

RaulFdzbis, 2019-11-14:
Just doing a fast reinstall (without even changing the configuration) solved the problem. Appart of that, I also had to delete some residual installation i had of openscenegraph, following this [issue](https://github.com/rdiankov/openrave/issues/417). 

So as a conclusion due to version changes try to do a fresh and **clean install** of openrave when trying this  [solution](https://github.com/crigroup/openrave-installation). Now is working :+1:

RaulFdzbis, 2020-01-10:
> For boost, you could work on [rdiankov/openrave#625](https://github.com/rdiankov/openrave/pull/625) (note that I have not tested that one)

Confirmed https://github.com/morxa/openrave/tree/boost-1.6x as of https://github.com/morxa/openrave/tree/e195ce9795b6ed7e3e4fe1c49a03ee16f83ca78d for https://github.com/rdiankov/openrave/pull/625 is compiling and working (openrave `v0.9.0`) on Bionic with the distro's Boost 1.65.

jgvictores, 2020-01-12:
Offtopic (`xenial`): Regarding [`master` at time of writing](https://github.com/rdiankov/openrave/tree/abe85928e74381d3d3d33db73693d92c9fbd7a1c), which identifies itself as `v0.25.0`, on a Xenial machine, I had to install https://github.com/Tencent/rapidjson from source because distro version missing `rapidjson/pointer.h`.

jgvictores, 2020-03-29:
Conversation already linked from other side, but specific comment (for future reference) on even newer Ubuntu versions: https://github.com/rdiankov/openrave/issues/756#issuecomment-604591630

lianghongzhuo, 2020-05-07:
I'm trying to Migrate from `boost::numeric` to` boost:numpy ` for Boost too. As this pull request's code is outdated. https://github.com/rdiankov/openrave/pull/625
In the current master branch https://github.com/rdiankov/openrave/commit/811cf2367a0ce8b6c92be26579ca161ab4bac0a8 I finished most cases by replace `numeric::array` to `numpy::ndarray`
But I stuck at here https://github.com/rdiankov/openrave/blob/master/python/bindings/include/openravepy/boostpython/boostpython_bindings.h#L23-L25
any suggestion on how to change these lines?

PeterBowman, 2020-09-09:
> Installing `rapidjson-dev` is not a complete fix since the `rapidjson/pointer.h` header file cannot be found afterwards: [ref](https://travis-ci.com/roboticslab-uc3m/openrave-yarp-plugins/jobs/252650010#L4295).

> Regarding [`master` at time of writing](https://github.com/rdiankov/openrave/tree/abe85928e74381d3d3d33db73693d92c9fbd7a1c), which identifies itself as `v0.25.0`, on a Xenial machine, I had to install https://github.com/Tencent/rapidjson from source (current [master](https://github.com/Tencent/rapidjson/tree/dfbe1db9da455552f7a9ad5d2aea17dd9d832ac1)) because distro version missing `rapidjson/pointer.h`

Documented at https://github.com/roboticslab-uc3m/installation-guides/commit/1e61468273a298dec65a2d095024e07cc40bbd29.

jgvictores, 2021-01-14:
Closed via https://github.com/roboticslab-uc3m/installation-guides/commit/a0a8b15256513105aeba095c598555ce3f2de3ac as well as https://github.com/roboticslab-uc3m/installation-guides/issues/81

closed 2021-01-14 by jgvictores