roboticslab-uc3m/installation-guides #67: [openrave] import openravepy

jgvictores, 2019-02-21:
In [1]: import openravepy
```
failed to import convexdecompositionpy /usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/convexdecompositionpy.so: undefined symbol: _ZN5boost6system15system_categoryEv
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-290d33d0afe4> in <module>()
----> 1 import openravepy

/usr/local/lib/python2.7/dist-packages/openravepy/__init__.py in <module>()
     33         openravepy_currentversion = loadstable(__openravepy_version__)
     34     except (NameError, AttributeError):
---> 35         openravepy_currentversion = loadlatest()
     36 
     37 if openravepy_currentversion is not None:

/usr/local/lib/python2.7/dist-packages/openravepy/__init__.py in loadlatest()
     14     return _loadversion('_openravepy_'+ver.replace('.', '_'))
     15 def loadlatest():
---> 16     return _loadversion('_openravepy_')
     17 
     18 def _loadversion(targetname):

/usr/local/lib/python2.7/dist-packages/openravepy/__init__.py in _loadversion(targetname)
     17 
     18 def _loadversion(targetname):
---> 19     mainpackage = __import__("openravepy", globals(), locals(), [targetname])
     20     global openravepy_currentversion
     21     openravepy_currentversion = getattr(mainpackage, targetname)

/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/__init__.py in <module>()
     39 from . import metaclass
     40 from . import interfaces
---> 41 from . import databases
     42 
     43 OpenRAVEModel = databases.DatabaseGenerator # for backwards compatibility

/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/databases/__init__.py in <module>()
    271 import grasping
    272 import convexdecomposition
--> 273 import linkstatistics
    274 import kinematicreachability
    275 import inversereachability

/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/databases/linkstatistics.py in <module>()
     70 from ..openravepy_int import RaveFindDatabaseFile, RaveDestroy, Environment, KinBody, rotationMatrixFromQuat, quatRotateDirection, rotationMatrixFromAxisAngle, RaveGetDefaultViewerType
     71 from . import DatabaseGenerator
---> 72 from .. import pyANN
     73 import convexdecomposition
     74 from ..misc import ComputeGeodesicSphereMesh, ComputeBoxMesh, ComputeCylinderYMesh, SpaceSamplerExtra

/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/pyANN.py in <module>()
     14 """Imports ANN extentions
     15 """
---> 16 from pyANN_int import *
     17 from pyANN_int import _pyann_exception_
     18 

ImportError: /usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/pyANN_int.so: undefined symbol: _ZN5boost6system15system_categoryEv
```

jgvictores, 2019-02-21:
Happens on Ubuntu 16.04 with OpenRAVE 0.15, fixed via downgrade to 0.9.

PeterBowman, 2019-04-21:
Adding `-lboost_system` did not help. See also https://github.com/rdiankov/openrave/commit/190c16557081daf8864cd119aaae20d7d6f87f07.

PeterBowman, 2019-06-16:
Reported upstream at https://github.com/rdiankov/openrave/issues/678. Suggested workaround:

```python
builtins.openravepy_version = '0.9'
import openravepy

# ...
```

...assuming that OpenRAVE v0.9 and v0.15 are installed alongside each other.

PeterBowman, 2019-12-14:
There has been another release to OR master branch (currently v0.25) in October which apparently resolves this issue.

closed 2019-12-14 by PeterBowman

PeterBowman, 2019-12-14:
BTW next OR release will allow compiling Python bindings with pybind11 (which seems to become the default option and thus disable Boost.Python).