roboticslab-uc3m/installation-guides #71: Document installation of PCan

PeterBowman, 2019-06-28:
Web: https://www.peak-system.com/fileadmin/media/linux/index.htm.

User manual: [PCAN-Driver-Linux_UserMan_eng.pdf](https://github.com/roboticslab-uc3m/installation-guides/files/3339541/PCAN-Driver-Linux_UserMan_eng.pdf)

To be linked from [specific-dependencies.csv](https://github.com/roboticslab-uc3m/yarp-devices/blob/fc417a504e1143f5f65e3e4d36027105fc2021a6/doc/specific-dependencies.csv). BTW also check [os-requirements.csv](https://github.com/roboticslab-uc3m/yarp-devices/blob/fc417a504e1143f5f65e3e4d36027105fc2021a6/doc/os-requirements.csv).

rsantos88, 2019-06-28:
If you want, I can document it with the steps I followed yesterday for the installation in the manipulation :point_up:

PeterBowman, 2019-06-28:
> If you want, I can document it with the steps I followed yesterday for the installation in the manipulation point_up

Sure, that would be great, thanks!

rsantos88, 2019-07-03:
https://github.com/roboticslab-uc3m/installation-guides/pull/72

rsantos88, 2019-07-03:
now it's ready to merge #72

PeterBowman, 2019-07-04:
Closed in https://github.com/roboticslab-uc3m/installation-guides/pull/72, thanks!

closed 2019-07-04 by PeterBowman