roboticslab-uc3m/installation-guides #77: [openrave] rapidjson not declared when installing on Bionic

RaulFdzbis, 2020-01-10:
Getting the same error as the one stated in the issue https://github.com/rdiankov/openrave/issues/707 when installing Openrave on Bionic.

Solved it installing **rapidjson**:

```bash
sudo apt install rapidjson-dev
```

Should we add this line to the installation guide for Bionic?

RaulFdzbis, 2020-01-10:
Possible duplicate of https://github.com/roboticslab-uc3m/installation-guides/issues/65, starting with comments on `rapidjson` at https://github.com/roboticslab-uc3m/installation-guides/issues/65#issuecomment-544455403.

jgvictores, 2020-01-10:
Added at https://github.com/roboticslab-uc3m/installation-guides/commit/8acedc6fe5148450380d55c07e0e32b59047ce41. Closing as duplicate of https://github.com/roboticslab-uc3m/installation-guides/issues/65.

PeterBowman, 2020-09-09:
Please note current OpenRAVE master branch requires a manual installation of RapidJSON from sources: https://github.com/roboticslab-uc3m/installation-guides/commit/1e61468273a298dec65a2d095024e07cc40bbd29 (https://github.com/roboticslab-uc3m/installation-guides/issues/65#issuecomment-689666111).

closed 2025-08-15 by PeterBowman