roboticslab-uc3m/installation-guides #78: Error compilando drivers JR3 en kernel 5

rsantos88, 2020-01-28:
Errores encontrados durante la compilación de los [drivers JR3](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-jr3.md) en Ubuntu 18.04.
La version de kernel que dispongo es la `5.3.0-26-generic` 
```bash
$ cat /proc/version
Linux version 5.3.0-26-generic (buildd@lgw01-amd64-039) (gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)) #28~18.04.1-Ubuntu SMP Wed Dec 18 16:40:14 UTC 2019
```
Al hacerlo me muestra el siguiente error:
```bash
$ make
make -C /lib/modules/5.3.0-26-generic/build M=/home/teo/repos/jr3pci-linux SUBDIRS=/home/teo/repos/jr3pci-linux modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-26-generic'
Makefile:227: ================= WARNING ================
Makefile:228: 'SUBDIRS' will be removed after Linux 5.3
Makefile:229: 
Makefile:230: If you are building an individual subdirectory
Makefile:231: in the kernel tree, you can do like this:
Makefile:232: $ make path/to/dir/you/want/to/build/
Makefile:233: (Do not forget the trailing slash)
Makefile:234: 
Makefile:235: If you are building an external module,
Makefile:236: Please use 'M=' or 'KBUILD_EXTMOD' instead
Makefile:237: ==========================================
  CC [M]  /home/teo/repos/jr3pci-linux/jr3pci-driver.o
In file included from /home/teo/repos/jr3pci-linux/jr3pci-driver.c:43:0:
/home/teo/repos/jr3pci-linux/jr3pci-ioctl.c: In function ‘jr3_ioctl’:
/home/teo/repos/jr3pci-linux/jr3pci-ioctl.c:111:57: error: macro "access_ok" passed 3 arguments, but takes just 2
         err = !access_ok(VERIFY_WRITE, (void *)arg, size);
                                                         ^
/home/teo/repos/jr3pci-linux/jr3pci-ioctl.c:111:16: error: ‘access_ok’ undeclared (first use in this function)
         err = !access_ok(VERIFY_WRITE, (void *)arg, size);
                ^~~~~~~~~
/home/teo/repos/jr3pci-linux/jr3pci-ioctl.c:111:16: note: each undeclared identifier is reported only once for each function it appears in
/home/teo/repos/jr3pci-linux/jr3pci-ioctl.c:114:64: error: macro "access_ok" passed 3 arguments, but takes just 2
                err =  !access_ok(VERIFY_READ, (void *)arg, size);
                                                                ^
/home/teo/repos/jr3pci-linux/jr3pci-ioctl.c:102:6: warning: unused variable ‘size’ [-Wunused-variable]
  int size = _IOC_SIZE(cmd); /* the size bitfield in cmd */
      ^~~~
/home/teo/repos/jr3pci-linux/jr3pci-driver.c: In function ‘jr3pci_init_module’:
/home/teo/repos/jr3pci-linux/jr3pci-driver.c:171:63: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  printk( KERN_INFO "jr3pci: JR3 PCI card detected at 0x%x\n", (int)jr3_base_address);
                                                               ^
scripts/Makefile.build:288: recipe for target '/home/teo/repos/jr3pci-linux/jr3pci-driver.o' failed
make[2]: *** [/home/teo/repos/jr3pci-linux/jr3pci-driver.o] Error 1
Makefile:1655: recipe for target '_module_/home/teo/repos/jr3pci-linux' failed
make[1]: *** [_module_/home/teo/repos/jr3pci-linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-26-generic'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
```

rsantos88, 2020-01-28:
La solución temporal que he aplicado en [jr3pci-ioctl.c](https://github.com/roboticslab-uc3m/jr3pci-linux/blob/develop/jr3pci-ioctl.c#L110-L114):

```C
#ifndef LINUX_20
	  if (_IOC_DIR(cmd) & _IOC_READ)
		      //err = !access_ok(VERIFY_WRITE, (void *)arg, size);
			err = 0;
	  
	  if (_IOC_DIR(cmd) & _IOC_WRITE)
	              //err =  !access_ok(VERIFY_READ, (void *)arg, size);
			err = 0;
```
Quitando esto, el driver ha compilado y se ha instalado correctamente.
Los LEDS de la tarjeta PCI encienden y arranca el device Jr3 de yarp. 
Dejo abierta la issue para arreglar el problema que da esta función con este kernel

jgvictores, 2020-01-30:
Could you verify if simply removing the first parameter is enough, as in https://github.com/martinezjavier/ldd3/commit/887600dab825111721c3bb9ce2331f28adbb648d ?

rsantos88, 2020-01-30:
Yes, it's enough

rsantos88, 2020-02-07:
I've fixed the error, detecting previously the kernel version (see https://github.com/roboticslab-uc3m/jr3pci-linux/commit/87536e37574b69cf6eb96dd04295bb27699f34fd)
Now it's working with the two Kernel versions. 
Note: I've removed `verify_area(VERIFY_READ, (void *)arg, size)` because it's deprecated and not used in our current versions.

rsantos88, 2020-02-07:
Note2: Added some installation notes if you want to install the drivers on Ubuntu 18.04 (https://github.com/roboticslab-uc3m/installation-guides/commit/68c47866e9f9bd63cd9bdc8daf2367488dd6babb)
I'll close the issue :)

closed 2020-02-07 by rsantos88

PeterBowman, 2020-02-07:
Not sure if we should care, but I presume that commit breaks compatibility with the Linux 2.0 kernel?