roboticslab-uc3m/installation-guides #81: Add Ubuntu 20.04 section to OpenRAVE install guide

PeterBowman, 2020-10-29:
I managed to launch our usual scripts (`teoSim` + some Python example at [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/b1560fa0afa8bf2a47f3b6822e297d0fee59afe1/examples/python)) by following these steps on Ubuntu 20.04 (Focal):

1. [Install dependencies](https://github.com/crigroup/openrave-installation/blob/1527ce0ace1d0ae620e91f43907b298aaafe20f7/install-dependencies.sh): a whole lot of apt packages + header-only RapidJSON ([install-rapidjson.md](https://github.com/roboticslab-uc3m/installation-guides/blob/d977c8f1a841283a8c6c59f7072a5f8f28f00d81/install-rapidjson.md)) + pybind11.
1. Install dev packages for OpenSceneGraph and FCL: `sudo apt install libopenscenegraph-dev libfcl-dev`
1. Clone latest OpenRAVE master. Tested at [4c43d49](https://github.com/rdiankov/openrave/tree/4c43d49895934b61cc286a6b41a10d3661fde465).
1. Configure:
   ```cmake
   cmake -DODE_USE_MULTITHREAD=ON \
         -DUSE_PYBIND11_PYTHON_BINDINGS:BOOL=TRUE \
         -DBoost_NO_BOOST_CMAKE=1 ..
   ```
1. Compile and install.

Most credits go to https://github.com/crigroup/openrave-installation. However, I did not build OpenSceneGraph and FLC from sources nor downgraded sympy to version 0.7.1. Not sure if this would cause me trouble later on.

For the record, it's not trivial to switch from Python2 to Python3 for OpenRAVE bindings. The build process stucks in the CMake configuration step:

```
(...)
-- MATLAB installation not found, is mex in the system path?
  File "<string>", line 1
    import sympy; print sympy.__version__
                        ^
SyntaxError: invalid syntax
-- failed to find python sympy system installation
-- System sympy (v=) is not right version, using local sympy
CMake Error at CMakeLists.txt:873 (install):
  install DIRECTORY given no DESTINATION!

-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so
-- Performing Test HAVE_ALL_PYTHON_HEADERS
-- Performing Test HAVE_ALL_PYTHON_HEADERS - Failed
CMake Error at python/bindings/CMakeLists.txt:50 (message):
  Should have installed pybind11; CMake will exit.

-- Configuring incomplete, errors occurred!
```

OpenRAVE embeds a few Python scripts within its CMake code for such things like checking the correct/expected sympy version. These scripts are not prepared for Python3 anyway, hence downgrading sympy as stated in several places is pointless.

jgvictores, 2020-10-29:
Sounds great!

Afaik:
- We'll have to look into `py3`. The depicted error looks simple enough to fix (from `print sympy.__version__` to `print(sympy.__version__)`
- Regarding downgrading `sympy`, unless they've done something about it, is required for certain OR database generators (ikfast for one, I believe).

Btw, the python bindings are nice to play around with, and also for the OR database generators, but they are not strictly a requirement for our setup/developments (e.g. `teoSim` became a bash one-liner a while ago).

jgvictores, 2020-11-09:
Was able to replicate up to here. I'd document 2 issues:
1. The previously mentioned Python-related comments. I believe upgrading can be quite a task, so I'd first investigate if a `virtualenv` setup is feasible.
2. A pretty big issue is regarding the absence of the `qtcoin` viewer, which afaik would leave us with no RGB camera. The most relevant old packages were `libsoqt4-dev` and `libcoin80-dev`. Focal has some similar packages, namely `libsoqt520-dev` and `libcoin-dev`; however, OpenRAVE does not recognize them as requirement fulfillments. What "worries" me most is that the new versions seem very tied to Qt5, whereas the old ones were strongly related to Qt4 (one thing is the fact that they can coexist on the same OS, which we have done in the past qt4->openrave and qt5->yarp, but another completely different issue is enforcing the qt5->openrave usage).

PeterBowman, 2020-11-10:
As you said, `libsoqt4-dev` does not exist on Focal ([ref](https://packages.ubuntu.com/search?keywords=libsoqt)). The apt manager suggests me to install `libsoqt520-dev` instead, which in turn is not available on Xenial nor Bionic...

jgvictores, 2020-11-24:
Notes to reproduce [initial comment](https://github.com/roboticslab-uc3m/installation-guides/issues/81#issue-732607145)

1. Install pybind11: `sudo apt install pybind11-dev`
2. Create simlink: `cd /usr/bin; sudo ln -s python2 python`

PeterBowman, 2020-11-24:
> Create simlink: `cd /usr/bin; sudo ln -s python2 python`

`update-alternatives` is your friend!

jgvictores, 2020-11-24:
Also, following the procedure, things are installed in `/usr/local/lib/python2.7/dist-packages/openravepy` but a simple `python` and `import openravepy` result in `openravepy_int`. @PeterBowman Perhaps you were using the examples that invoked OpenRAVE via Python bindings of YARP and not the actual `openravepy` module?

PS: My objective is to reach some functional steps using the sympy downgrade, but within a Python venv or similar to not affect at system-level.

PeterBowman, 2020-11-24:
> Perhaps you were using the examples that invoked OpenRAVE via Python bindings of YARP and not the actual `openravepy` module?

Nope, I do have all necessary binaries compiled and installed, I presume. In my case, `python2 -c 'import openravepy'` throws no error:

``` bash
~/git/openrave/build (master % u=)$ cat install_manifest.txt | grep python
/usr/local/share/openrave-0.54/cppexamples/orpythonbinding.cpp
/usr/local/lib/python2.7/dist-packages/openravepy.egg-info
/usr/local/lib/python2.7/dist-packages/openravepy/__init__.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/metaclass.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/openravepy_ext.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/misc.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/trajectoryutils.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/pyANN.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/__init__.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/tutorial_plotting.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/testupdatingbodies.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/checkconvexdecomposition.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/tutorial_grasptransform.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/testphysics_diffdrive.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/tutorial_iktranslation.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/tutorial_iksolutions.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/tutorial_inversereachability.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/inversekinematicspick.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/tutorial_iklookat.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/hanoi.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/constraintplanning.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/simplemanipulation.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/qtserverprocess.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/testviewercallback.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/fastgrasping.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/calibrationviews.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/fastgraspingthreaded.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/testphysics.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/graspplanning.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/tutorial_iklookat_multiple.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/simplegrasping.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/collision2.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/visibilityplanning.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/pr2turnlever.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/tutorial_iktranslation2d.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/collision.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/__init__.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/qtexampleselector.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/dualarmdemo_schunk.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/showsensors.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/simplenavigation.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/testphysics_controller.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/checkvisibility.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/tutorial_ik5d.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/cubeassembly.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/examples/movehandstraight.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/interfaces/visualfeedback.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/interfaces/TaskManipulation.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/interfaces/Grasper.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/interfaces/__init__.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/interfaces/BaseManipulation.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/databases/visibilitymodel.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/databases/kinematicreachability.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/databases/grasping.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/databases/linkstatistics.py
[... cut ...]
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/ikfast_generator_cpp_sympy0_6.py
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/openravepy_int.so
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/pyANN_int.so
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/convexdecompositionpy.so
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_54/openravepy_configurationcache.so
```

Here is my [CMakeCache.txt](https://github.com/roboticslab-uc3m/installation-guides/files/5591339/CMakeCache.txt).

jgvictores, 2020-11-25:
Thanks for all the info!
Actually fixed my installation via the missing python packages line (https://github.com/crigroup/openrave-installation/blob/1527ce0ace1d0ae620e91f43907b298aaafe20f7/install-dependencies.sh#L44). :tada:

jgvictores, 2021-01-14:
Closing issue as current state has served well to install OpenRAVE on over 20 machines, with diverse Bionic and Focal setups.

For the record, these are the two aspects where we could further optimize, but closing in current state due to the reason state above:
- Recommend using `apt` version of OSG on Focal, skipping `./install-osg.sh` which compiles from source (and clashes if the apt package is installed)
- Recommend not using FCL at all, skipping `./install-fcl.sh`.

closed 2021-01-14 by jgvictores