roboticslab-uc3m/installation-guides #83: teoSim errors during execution

rsantos88, 2021-05-18:
I've installed Openrave on Ubuntu Focal Fossa distribution following the instructions documented in [install-openrave-0540-ubuntu-2004-focal](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-openrave.md#install-openrave-0540-ubuntu-2004-focal) + [teo-openrave-models](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/master/doc/teo-openrave-models-install.md#install-the-models) + [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/master/doc/openrave-yarp-plugins-install.md#install-openrave-yarp-plugins-on-ubuntu-working-on-all-tested-versions) + [Boost](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-boost.md#install-boost-ubuntu) and it apparently works but the following errors are displayed in the terminal:

[image]
[teoSim-terminal.txt](https://github.com/roboticslab-uc3m/installation-guides/files/6502935/teoSim-terminal.txt)

PeterBowman, 2021-05-18:
Those are warnings. Do they prevent teoSim from running for you? I'm already used to them, they're just bloating the terminal, everything works fine.

rsantos88, 2021-05-18:
Perfect. If all this is normal, I close the issue.
Thanks!

closed 2022-05-23 by PeterBowman